ButterFlight/butterflight #20: Motolab tempest F4

mtnmikev, 2018-03-01:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

mtnmikev, 2018-03-01:
Using crossfire for tx/rx

kidBrazil, 2018-03-01:
Just waiting on @mtnmikev to confirm that the test hex I provided him indeed fixes his issue - Don't have Motolab Tempest F4 to test agains.

kidBrazil, 2018-03-08:
@mtnmikev My friend, I am giving you until Monday to report back on this issue or we are going to close it.

kidBrazil, 2018-03-18:
User gave up - no testing ever happened

closed 2018-03-18 by kidBrazil