ButterFlight/butterflight #25: CPU hangs at any overclocking rate with Butterflight 3.4.1: Overclocking bug Butterflight

jstremmler, 2018-03-04:
FC OmnibusF4 Corner: 

- Butterflight 3.4.1, CPU Overclocking = 215MHz , result CPU hangs; 
- Betaflight 3.3.0 CPU Overclocking= 215MHz CPU works at 50%. with same FC
- both fimrware were operating  with 32/16kHz Gyro/PID sampling rate 
- without CPU overclocking Butterflight 3.4.1 shows a CPU load of 83% with 32/16k sampling rate
- CPU hangs at any overclocking rate with Butterflight 3.4.1

dzenbot, 2018-03-05:
Would this explain why being able to arm would take up to 20-30 seconds of delay, after powering the FC?

kidBrazil, 2018-03-08:
Ornery has merged a fix that should solve the lockup issue.

closed 2018-03-08 by kidBrazil

dzenbot, 2018-03-08:
Which is now available in https://github.com/ButterFlight/butterflight/releases/tag/3.4.2 YAY