ButterFlight/butterflight #27: vtx_akk_hack doesn't seem to support RDQ Mach 2

TheAdamizer, 2018-03-04:
Noticed that whatever I tried I couldn't get the legacy hack to work with my racedayquads mach 2 vtx using a known good configuration (on my omnibus pro corner f4).  I was also using softserial to get it working but this wasn't a problem before.

Reproducible by attempting to make the akk_hack setting work with any (i'd guess) mach 2 vtx and attempting smart audio control via osd.  *sometimes* you can actually see the menu to choose channel / power level but setting those values doesn't actually change the vtx.

This was on the 3.4.1 release of butterflight.

I know this bug is not hardware related because I forked butterflight, checked it out, ran
`git revert -m 1 30f3152`
(this is the merge commit for the betaflight pr that breaks mach 2 compatability)
built this and flashed it to my pro corner and it worked like a charm even over soft serial.  I of course ran into some conflicts but I always chose the incoming changes in those 3 cases and had no trouble building.  Of course I may have broken some stuff like setting power level on disarm and setting custom frequency but I think it's evidence enough this feature may require some more attention.

here's my diff:
```# diff

# version
# ButterFlight / OMNIBUSF4SD (OBSD) 3.4.1 Mar  4 2018 / 02:00:33 (1cd8f73) MSP API: 1.38

# name

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# mixer

# feature
feature SOFTSERIAL

# beeper

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set rssi_channel = 6
set serialrx_provider = SBUS
set osd_vtx_channel_pos = 2282

# profile
profile 0

# rateprofile
rateprofile 0

# ```

TheAdamizer, 2018-03-04:
Here is the reverted fork I built that seems to work perfectly with mach 2s: https://github.com/TheAdamizer/butterflight

kidBrazil, 2018-03-05:
@TheAdamizer - so what you are saying is that setting VTX_AKK_HACK to ON didn't work for you?

TheAdamizer, 2018-03-05:
@kidBrazil I'm saying that not only did it not work for me, but that there is a functional difference between the old implementation of smart audio and the new implementation of smart audio with the akk hack such that the old implementation works and the new implementation does not, so hopefully something more specific than simply "vtx_akk_hack didn't work for me"

TheAdamizer, 2018-03-05:
Probably at least part of the problem is that you're not conditionally setting SA_QSIZE to 4 given VTX_AKK_HACK is on which could be something that the mach 2's rely on.  I'm not a C programmer at all however so I didn't try diving further into it.

ijustwannafly, 2018-03-05:
Here are the changes required 

https://github.com/krankenreif/betaflight/commit/90efdec9da7722a74121758379a004dec38e9146

Based on the existing code of butterflight something along the lines of:

if (vtxSettingsConfig()->akkStyleEndFrame)
{
     #define SA_QSIZE 4
}
else
{
     #define SA_QSIZE 6
}
---------------------------------------------------------
if (vtxSettingsConfig()->akkStyleEndFrame)
{
     portOptions_e portOptions = SERIAL_STOPBITS_1;  
}
else
{
     portOptions_e portOptions = SERIAL_STOPBITS_2; 
}

Now, I'm not a good coder and this is based off the last time I coded anything nearly 15 years ago.

But I'm sure any good coder would understand what I am setting forward given the information.  I have no ability or time now to compile or create a build environment.  Good luck.

TheAdamizer, 2018-03-05:
Yea like I said I'm not a C dev I write JS lol so I tried to do something exactly like that but got some weird error about some constant being variable at runtime (duck typing for life, yo lol) and was like yea...I'm not gonna write reliable code here so I'm just gonna revert and get my own quad in the air.

kidBrazil, 2018-03-05:
@TheAdamizer Thanks man, I appreciate you giving me the details - I will look into it since I did the implementation and get this resolved ASAP.

@ijustwannafly - thank you for that!

kidBrazil, 2018-03-06:
@TheAdamizer - @LexioTech has been working on a new implementation for the AKK thing using features. He will be adding the SA_QSIZE and other missing parameters

kidBrazil, 2018-03-08:
Closing issue since the fix has been merged into master

closed 2018-03-08 by kidBrazil