ButterFlight/butterflight #39: Multishot causing motors to spin up

MrNoizeFPV, 2018-03-07:
so have an issue I usually run dshot but having some weird issues with it on hard front flips, so put multishot on plugged in the battery and 3 of the motors started spinning up slightly not enough to cut your fingers but sketchy enough that I had to put my foot on the quad to unplug it very dangerous im on the helio board as well have not had the issue on dshot

flint723, 2018-03-08:
In Config tab change your "minimum command" under multishot to like 980 (or lower if you have to) instead of 1000.

kidBrazil, 2018-03-08:
@MrNoizeFPV can you please try @flint723 suggestion and get back to us?

papayou66, 2018-03-08:
@MrNoizeFPV did you calibrate your mutlishot from motor TAB ? 1000-2000

MrNoizeFPV, 2018-03-08:
Hey guys sorry just seeing the mesages I lowered the command and that did it worked fine but Im seeing a weird what seems like a desync issue on dshot and multishot ive played with the idle values for both to see if that fixes it, but when I do a hard front flip the quad does not commit to the flip and kicks back im able to save it every time and it does not happen every time, part of me feels like it could be me when I flick the stick fast for the flip its like im bringing the stick back slightly and not committing its weird but I have had it happen on ms and dshot now. also wondering if its the loop speed running 16/16

MrNoizeFPV, 2018-03-08:
papayou no I calibrated in blheli

papayou66, 2018-03-08:
wich value ? and i think it's better to make a manual calibration from the GUI

MrNoizeFPV, 2018-03-08:
papayou check my long message above i got it to work fine no more spinning motors my issue im having now is a weird one

kidBrazil, 2018-03-08:
@MrNoizeFPV Can you record a BB of the weird issue and send it to us please?

MrNoizeFPV, 2018-03-08:
@kidBrazil just sent you a bb in butter slack

kidBrazil, 2018-03-09:
After reviewing @MrNoizeFPV it was found that he had RC Interpolation setup incorrectly and therefore there is no bug to be fixed.

closed 2018-03-09 by kidBrazil