ButterFlight/butterflight #40: F7 Flight Controller Maxing out CPU Load

SupaCoop, 2018-03-08:
Hi, I recently attempted to make the switch to butterflight and am having issues running 32khz sampling. I'm running the betaflight f7 on 3.4.2 with the icm20602 gyro enabled and it reads 100% cpu load on 16khz/16khz without the dynamic filter. I've seen guys with f4's running 32khz/16khz with the dynamic filter enabled and they were running just fine. I tried to contact fpv model and they literally told me to wait for the next firmware version to come out. An f7 should have plenty of processor cycles to spare even with 32khz/32khz and the dynamic filter, right? Any ideas?

Zuldan, 2018-03-08:
@SupaCoop I'm glad you raised this issue. I was going to do it today myself. My Matek F722-STD F7 is having the same issue. Unfortunately it's a known issue. The Betaflight code is optimised primarily for F3/F4 FC's. I think the devs of Betaflight are planning to bring F7 optimisations in v3.4. The Kalman filter works best on 32khz/32khz and to run that without overclocking you need a F7. Hopefully the Butterflight devs will merge those optimisations or maybe make better optimisations.

@Butterflight devs thank you for all the hard work you're doing and focusing on what the community wants.

SupaCoop, 2018-03-08:
@Zuldan Sounds good, thanks so much for the info man!

closed 2018-03-10 by kidBrazil

wesk-ms, 2018-03-10:
I see this issue is closed, did it get fixed?  No PR has been referenced.  Would like to try Butterflight, so I was just checking.

Just FYI, this issue appears to have been fixed now in betaflight.  On a MatekF722, I can run 32K/16K and my CPU utilization sits around 20%.

PR #5237 in build #646 fixes the CPU utilization issue...
https://github.com/betaflight/betaflight/pull/5237

Also, F7 overclocking is enabled by betaflight PR #5372 in build #656
https://github.com/betaflight/betaflight/pull/5372

SupaCoop, 2018-03-10:
I guess it’s fixed now if the betaflight devs fixed it. I didn’t close the issue I just got an email saying it closed. I will try those builds, thanks so much for the info!

SupaCoop, 2018-03-10:
@wesk-ms I just tried the latest build #662, and it still locks up at 100% cpu load. I even had it on 16K/8K with dynamic notch off.

wesk-ms, 2018-03-10:
BF 662 backs out some recent dshot changes that had issues on some F7 targets.

I just flashed and configured BF 662 on my Matek F722 and, for me, it's the best build so far for the F7....on the bench.  Will fly it in a little while to see how it does in the air.

Attached a screenshot showing CPU utilization at 9% with 16k/16k
[image]
.  Very surprised at how good things are.

@SupaCoop if you attach a 'dump' file, I can compare it with mine.  Also, make sure you reset to defaults after flashing by typing 'defaults' in the CLI before you start setting the new build up.

wesk-ms, 2018-03-10:
@SupaCoop  On BF build 662, 32k/32k does cause 100% cpu utilization for me.  32/16k maxes out at 27%, though.  And that's with a GPS module I just added to experiment with.

[image]

SupaCoop, 2018-03-11:
@wesk-ms Here's the dump file man, hope this helps!

[BTFL_cli_ dumpfile_20180311_000819.txt](https://github.com/ButterFlight/butterflight/files/1800068/BTFL_cli_.dumpfile_20180311_000819.txt)

SupaCoop, 2018-03-11:
@wesk-ms Sorry, didn't have 32k enabled in first dump file... This one does!

[BTFL_cli_dumpfile_20180311_002044.txt](https://github.com/ButterFlight/butterflight/files/1800071/BTFL_cli_dumpfile_20180311_002044.txt)

wesk-ms, 2018-03-11:
Interesting.

I thought you had a Matek F7.  The BF F7 has two gyros, where the Matek only has one.  I didn't see any settings that would explain the difference in CPU utilization that you are seeing.  Overclocking your board to 240mhz would help a little, but not near enough to make it flyable if you're pegged at 100% with the near vanilla config you attached below.

I've got a couple of the Omnibus F7 V2s and had a problem similar to what you're seeing.  They also have the dual gyro setup.  The resource IO list is surprisingly different on the dual gyro boards compared to the Matek F7.  I really think the issue is the F7 processor implementation used on the BF boards.  Could be on all the dual gyro boards, but won't know that for sure until I power up one of my Omnibus F7s.

Well, at least you should be able to rule out your configuration, now.  Switching to a Matek F7 for ~$35 sure seems like a cheap way to move on.  At least until the betaflight and butterflight devs get the F7 issues fully worked out.

SupaCoop, 2018-03-11:
@wesk-ms Alright, sounds good. Thanks so much for your time and help, I really appreciate it!