ButterFlight/butterflight #43: MMIX and Motor Resource Not Set Correctly for Reverse Prop Direction and 180 Degree Board Alignment

f4ichuck, 2018-03-08:
1. Set your board and sensor alignment 180 degrees and select motor direction reversed.
2. You will not be able to arm your flight controller and motors are still in default order if you use the motors tab to spin each motor.
3. Using Racerstar Tattoo F4S AIO (target is LUX4OSD)
4. Flight controller info, identifier: BTTR, version: 3.4.2
5 Workaround is to manually assign motor resources and add mmix shown below.

# diff

# version
# ButterFlight / LUXF4OSD (LUX4) 3.4.2 Mar  7 2018 / 18:01:10 (2dc6f02a5) MSP API: 1.38

# name
name QAV-R 220

# resources
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 B01
resource MOTOR 4 B00

# mixer
mmix 0  1.000 -1.000  0.750 -1.000
mmix 1  1.000 -1.000 -0.750  1.000
mmix 2  1.000  1.000  0.750  1.000
mmix 3  1.000  1.000 -0.750 -1.000

# feature
feature ANTI_GRAVITY

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1700 2100 0
aux 1 13 0 1300 1700 0
aux 2 28 1 1700 2100 0
aux 3 35 0 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW180
set gyro_stage2_lowpass_hz = 150
set gyro_filter_q = 19
set gyro_offset_yaw = -180
set align_acc = CW180
set acc_hardware = NONE
set align_mag = CW180
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 11
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = -180
set bat_capacity = 1500
set vbat_scale = 109
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_vbat_pos = 2433
set osd_rssi_pos = 2401
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2454
set osd_flymode_pos = 333
set osd_throttle_pos = 2424
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2439
set osd_mah_drawn_pos = 2447
set osd_craft_name_pos = 2057
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_endbatt = ON
set osd_stat_tim_1 = ON
set cpu_overclock = 192MHZ

# profile
profile 0

set dterm_notch_hz = 0
set anti_gravity_gain = 6000
set p_pitch = 60
set d_pitch = 28
set p_roll = 47
set d_roll = 27

# rateprofile
rateprofile 0

set roll_expo = 12
set pitch_expo = 12
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 78

#

Zuldan, 2018-03-17:
> sensor alignment 180 degrees and select motor direction reversed

Thanks for the heads up. These are the exact two settings I use. My build is still in-progress. Glad I hadn't done a test flight yet lol. Hopefully the fix for this will be in the next release.

kidBrazil, 2018-04-12:
@f4ichuck - Still an issue with latest builds?

closed 2018-04-12 by kidBrazil

f4ichuck, 2018-04-14:
I haven't tried it yet!