ButterFlight/butterflight #48: Please put back "disarm regardless of throttle value" option

tavdog, 2018-03-09:
I need this feature for the Matek F405 to prevent mid-air disaarms due to errant rapid disarm/rearm events as seen in blackbox logs.

halfcab1981, 2018-03-10:
Yes please! Another vote for this here. Even if just in the CLI

chriswils, 2018-03-10:
Can't this be fixed by using the timeout? Guessing your disarm/arm happens quite quickly?

kidBrazil, 2018-03-20:
wait.. so you guys want us to add back the option that does not allow the quad to be disarmed if the throttle value is above "0"?

halfcab1981, 2018-03-20:
Yes please! My typical landing workflow is that I approach the landing zone and when I'm a few metres away I flick the disarm switch but the quad doesn't Disarm because of this option. I bring the quad down to the grass to land by reducing the throttle gently to zero. 

In the event of an emergency, I've trained myself to Disarm by flicking the switch and reducing the throttle  to zero simultaneously. It doesn't slow down the Disarm at all. I do agree that people new to the hobby probably shouldn't use it though

I really like the added safety factor of not being able to accidentally Disarm by bumping the Disarm switch when flying. I know it's an unlikely situation but if you're flying over water it's one little safety feature that could prevent a lost quad

kidBrazil, 2018-03-20:
@halfcab1981 I am not sure I agree with your reasoning. In my opinion it is MUCH safer to have the quad disarm instantly especially on racing scenarios. No matter how well you have trained to throttle zero many have not and that extra time might be crucial in preventing injury. 

You should be able to learn how to land by gradually throttling down and then just disarming 1" or less off the ground.

Now for cases like @tavdog - this seems like a bandaid for a different issue. Perhaps we should look at WHY you are getting the constant arm/disarm. Have you checked to make sure your radio is in good order?

halfcab1981, 2018-03-21:
I'm perfectly capable of landing by gradually throttling down and then just disarming 1" or less off the ground. I can do it both ways, I'm just saying I prefer my method personally and for me there is no downside. I certainly understand that it should not be a default or indeed even available in the GUI but to remove it completely is to deny those that use it and have done so for many years the choice.

Look at aircraft cockpit design. Any switch that has a good result in serious consequences of unintentionally activated, like the fire extinguisher switches, has a physical guard so it can't be unintentionally switched. Same thing here. Pilots are trained that in an emergency they raise the switch guard and then activate the extinguisher. It is perfectly reasonable that some people would like this on their quad. 

Just as some quad pilots say they have never accidentally disarmed, I have never had a situation where I was unable to disarm because of this option. Having it as a CLI option that can only be turned on if you really want it, would be great! But obviously not as a default, and not for inexperienced pilots.

libwtf, 2018-03-22:
@halfcab1981 why not set that up in openTX?

kidBrazil, 2018-03-23:
Closing the issue since we have decided not to change this. Being able to disarm the quad immediately without having to move the throttle is a safety thing. What if you cant move your throttle down for whatever reason?

I suggest that for those who are accidentally disarming their quads that perhaps they look at using a double arm switch or move it somewhere you are less prone to hitting it accidentally.

To the chap with the weird ARM/DISARM Blackbox issue.. I would look deeper into your hardware to figure out what is causing that issue.

closed 2018-03-23 by kidBrazil

iforce2d, 2018-05-03:
It was always only an option, and the original poster is just requesting that it remain just that - optional for those who want it, like me. It need not affect anybody else if they don't like it. But good luck with your 'my way or the highway approach' to development, I hope it works out.

quiljw01, 2019-05-09:
I don't think devs should remove a feature that people actively use safely.  Very annoyed by this.  Please add the feature back.  If you think it's a safety issue then just leave it in CLI only, don't put it in the GUI like it was originally.   You're right, it probably shouldn't be in the GUI in the first place, it should only be for experienced users who know how to find it and how to use it.  

For @kidbrazil , what happens if for whatever reason your disarm switch can't go back to disarm position? Same thing.  

I should start compiling my own versions of betaflight with this feature and sell them for $1.

quiljw01, 2019-05-09:
@libwtf some people don't use opentx , most simple radios cannot do this.