ButterFlight/butterflight #49: Wrong Cell count reading and voltage reading Matek F405

mawais15, 2018-03-10:
I flashed latest Butterflight 3.4.2 Matek F405 target upgrading from betaflight 3.3 and found out that the cell count in the ADC voltage sensor is wrong. It is showing 7 cells when connected 4S battery and 5 cells when connected with 3S. Voltage readings are wrong and current scaling provided by Matek also yields wrong current readings. 
[https://ibb.co/g72cYS](url)

halfcab1981, 2018-03-13:
I am getting the same thing on my DYS F4. Getting 5s showing as a low voltage 4s. I had to increase my max voltage to rectify.

kidBrazil, 2018-03-20:
@mawais15 @halfcab1981 have you tried adjusting voltage scaling?

halfcab1981, 2018-03-20:
Not yet but I will now. Is it possible for this to have changed across various versions of betaflight and now butterflight? Because it worked normally without any changes to defaults in the past

kidBrazil, 2018-03-23:
@halfcab1981 it is possible that something got changed on the defaults. Please try adjusting your voltage scaling and see if that resolves it. Report back your value and we can go from there.

frakkan, 2018-03-23:
I am using Matek F405 on 6 of my quads, none of them has any issue with
voltage.

11Frontier, 2018-03-25:
Any chance of just adding in the option to put in your cell count?

wt29, 2018-04-15:
Same issue F405 AIO shows 5 cells on a 4S Battery. Have disabled to battery beep ATM

Ryano12, 2018-04-22:
This is not a bug? Just adjust your scaling so that your voltage is correct.
There is no need for an option to manually select battery cell count. As the voltage ranges don't intersect for different cell counts (eg. A fully charged 4s still has lower voltage than a dead 5s)

Sm1thson, 2018-05-08:
I currently have Betaflight on my F405 rig and I had this issue, to resolve I (props off) plugged in a freshly charged battery whilst connected to the configurator then adjusted the voltage scaling so it read 4.2V. 

so not a biggie, but yeah Ive never had to mess with the voltage scaling before.

kreidler, 2018-05-19:
It is not a Butterflight issue. Same is on Betaflight.
Most use the standard setting of 110. If you increase to 112 you will get with (good) batteries one cell more detected.
For me it helps to double the accuracy and then go one step back from 224 to 223:
```
set vbat_scale = 223
set vbat_divider = 20
```

11Frontier, 2018-05-25:
Betaflight return invalid name when using

set vbat_divider = 20

kreidler, 2018-05-28:
@11Frontier : You are right. CLI does not support vbat_divider in Butterflight. Butterflight configurator supports vbat_divider. So, please use the configurator yet.

@kidBrazil : As I am not a programmer - Could you please add vbat_divider to CLI.

orneryd, 2018-06-07:
this is coming after we finalize the release 3.5.1. I have already merged it up into the imuf_107_odin_bef_rebase branch from BeF.

closed 2018-06-07 by orneryd