ButterFlight/butterflight #50: Kakute F4 V2

mrkevvzime, 2018-03-10:
I armed and quad rocketed out... it lost battery immediately so log is very short.
[BTTR_BLACKBOX_LOG_20180310_180634.zip](https://github.com/ButterFlight/butterflight/files/1799662/BTTR_BLACKBOX_LOG_20180310_180634.zip)

kidBrazil, 2018-03-20:
can you carefully try it at 16/16 - try arming it without props on first and see how the motors behave.

Then try it again safely from a distance with props on. Make sure props are on properly and your motors are spinning the right direction.

ABLomas, 2018-03-24:
I tried similar setup today. Started with 32/16 (with overclock) - immediately after arm quad goes up, because it oscillates on roll axis. 10s and motors will be cooked.
D-term overshoots a lot on blackbox logs. I lowered pid's to very low values, lowered D to 10 or so - no difference. Even without props, on desk motors are creating high-pitch sound, with static quad - D-term is way off.
Lowered to 16/16 - better, but still happens with D > 20.
8/8 - looks good, could not reproduce on bench.
I have blackbox logs but can't attach them (zip file - still says that "we don't support that file type"). I also used butterflight with 32/16 on other FC's (Matek F405, Omnibus F4 with the same gyro) and they work fine. Last time i have seen something similar - when betaflight 2.8 or smth was released with complete filters rework (and everyone needed to dampen FC's and motors, and lower "D").

mrkevvzime, 2018-03-26:
When it's set 8/8 it flies but there are terrible oscillations. I saw a YT video with the same problem and it was fine with 8/4 settings. I have to try it out. When i set 32/x it goes mad...

ABLomas, 2018-03-26:
I tested it yesterday, flies fine on 8/8, no problems.
Unusable on any higher gyro readings (16, 32K). I checked motors, props, they are smooth (CMOS FPV cam (eagle2) do not show jello), also kakute has "integrated" dampener.

kidBrazil, 2018-03-28:
A lot of work is going into 32khz right now so a little patience will be needed. For now I would say run it at 8/8 until we can sort it out.

ChristAndQuads, 2018-04-08:
I changed to Multishot and that fixed this issue for me for now. I think it may be the ESC firmware not working with DSHOT well.

Ringo-gph, 2018-04-09:
I'm using Kakute F4 V2 with Tekko32 4in1. I've got it working using latest ButterFlight on 32/16, dynamic filters, PT1, all filters turned off, acceleromter turned off, overclocked to 216Mhz with a CPU load of about 43% (blackbox off, not sure if it eats up CPU, but I'm sure it does.)

There is one issue I have and it caused death rolls when I first put it together and tried to fly. I took props off and in the motors tab when I tried to hit "HOME" and then "END" on each motor, two motors would cause the FC to lock up requiring me to unplug the lipo and the USB cable to reset it. I finally fixed this issue by turning the Tekko32 4in1 over.

I've built a second one and tried to keep the Tekko32 4in1 oriented the original way that it comes in the stack sold by Holybro. I thought I had it working but motor 1 (really ESC pad M3 as I rotated it around so the lipo leads come out the back) will cause it to lock up hitting "HOME" and then "END", but if you don't aggressively ramp up the power on it, it's fine. I took it for a test flight though and it eventually fell out of the air and even when I tried to disarm it, the results screen on BF didn't show up which tells me that the FC locked up again due to the motor.

I'm going to flip the ESC over later on (most likely tonight) and try it again.

I thought I'd share here since I see others cannot get it running on 32/16, but I also wanted to share the weird issue I'm having. I'll will try 8/8 with my overclock and then 8/8 with no overclock and see if I still have the same issues to determine if it's the loop time (I doubt it's the overclocking issue ...and I stay below 40C with my other one overclocked to 216)

SnappyFPV, 2018-04-16:
Same issue ... was a faulty Tekko32 35A ... 1 motor Out was desyncing motor and smoking

orneryd, 2018-06-07:
is this working in BeF? I have rebased as of this morning into the imuf_107_odin_bef_rebase branch