ButterFlight/butterflight #52: Debug_Mode=Notch Reporting as Airmode in Cleanflight/Betaflight BlackBox Explorers

spatzengr, 2018-03-12:
Butterflight logs that are set to record a debug mode = Notch, is being recognized as Airmode in BBE.  What I don't understand is that it works fine in Betaflight 3.3 RC1, RC2, RC3 and Stable for which I understand Butterflight was forked off sometime during that time period.  One potential:  I'm noting that the logs are being recognized by BBE as Cleanflight logs vs. Betaflight logs.  Maybe if that were changed the downstream would work correctly?  If my memory is serving me well, at some point, Notch and Airmode positions were swapped - or general reordering, etc... - made some old Betaflight logs do the same.  But it was fixed in BBE, but maybe only or logs recognized as Beaflight logs?

See the attached logs for an example of the issue.  Also note this is occurring with Cleanflight BBE 2.5.11, Cleanflight BBE 3.0.0-rc1 and Betaflight BBE 3.0.0-rc1.

[Blackbox.zip](https://github.com/ButterFlight/butterflight/files/1802860/Blackbox.zip)

kidBrazil, 2018-03-14:
Thanks @spatzengr we will be looking into the issue.

brucesdad13, 2018-03-15:
Debug modes are based on a enumerator and differ between releases and forks. Probably need a Butterflight specific fork of the Blackbox explorer also.

spatzengr, 2018-03-15:
Can't it be baked into the existing BBE, like it is for Cleanfligh, Betaflight, and iNAV?

Seems like a better use of human resourse vs. splitting for such a utility tools that doesn't get that much attention as it stands.

kidBrazil, 2018-03-15:
@spatzengr @brucesdad13 - Thanks for the info I think I will just fork BBE so I can make a PR to include BTTR as one of the enumerators. THat way we wont default to Cleanflight.

wesk-ms, 2018-04-14:
@kidBrazil, fwiw, I'm unable to open any Butterflight logs in the 10.4.0 configurator that have any debug settings enabled.  Disable the debug settings in the CLI, go fly, and those logs open just fine.  I'm using 3.5.0 on a Matek F722.  Was trying to use the notch debug settings to fine tune some things.

When a debug level is configured and you try to open the log, the configurator becomes unresponsive and the log never opens.  No error message is displayed and you have to close the app to switch back to the configurator.

wesk-ms, 2018-04-14:
Update:
Looks like this is the root cause of BBE hanging when opening certain logs:
[#5625] (https://github.com/betaflight/betaflight/issues/5625)
[#5645] (https://github.com/betaflight/betaflight/pull/5645)

spatzengr, 2018-04-14:
@kidBrazil , now.  It was working fine for the 1st edition of the configurator that had the built-in BBE.  I have not checked subsequent versions of the configurator.

orneryd, 2018-06-07:
this is coming after we finalize the release 3.5.1. I have already merged it up into the imuf_107_odin_bef_rebase branch

closed 2018-06-07 by orneryd