ButterFlight/butterflight #53: Feature Request: Map radio auxiliary channel to general purpose output

totterfree, 2018-03-12:
Problem:
   Our flight controllers don't have a way to communicate with "dumb" hardware. There is no software for controlling generic outputs on our flight controllers (like gpio on arduino). We have PWM, but there are many use cases which require a simple high/low state.

My Specific Use Case:
   I would like to be able to better control the on/off state of my video transmitters. Reason being, that I (and many others) need to be able to compete in team races where quads are not recoverable after crashing. Currently SA is not robust enough to be powered off on initial plugin, powered on via aux, and powered completely off in the event of a downed (but not unplugged) quad. In order to solve this, I would like to be able to switch the power to my VTX on and off via something like a transistor. The only problem is that I have no way (using cleanflight, betaflight, raceflight, or butterflight, or to do so.

Proposed Solution:
   I propose a generic purpose output resource which could be mapped to UART, buzzer, LED, etc in order to control such circuits. This is obviously dependent on if those pins are capable of outputs other than PWM.

Note:
   It's possible such functionality might also include generic purpose input as well, in which case we are really just talking about mapping a GPIO pin. Since this is not part of my use case I am not including it in this feature request, but it's worth considering from a developer standpoint.

chriswils, 2018-03-12:
I second this. All pins that can do pwm can also output simple on/off (or just 0-100% duty cycle).

joeostrander, 2018-03-12:
I think PinIO/PinioBox is in this fork -- I've been meaning to try it...
https://github.com/betaflight/betaflight/wiki/Pinio-and-PinioBox
https://github.com/betaflight/betaflight/issues/5338

joeostrander, 2018-03-13:
FYI, I just tested PinIO and it worked great.

totterfree, 2018-03-13:
@joeostrander that's good to hear! Now that's only indicating the current flight mode correct? Or is it something that can be mapped to an radio aux channel?

joeostrander, 2018-03-13:
It maps to radio aux channel.  After entering CLI commands, you'll get new box(es) like *User 1* on the Modes tab.

kidBrazil, 2018-03-14:
@totterfree is this issue then resolved by the pinIO?

totterfree, 2018-03-15:
@kidBrazil It looks like it is indeed! I'd like to give this a test before saying it is for sure. Hopefully this week I can confirm. Is it likely that we could merge the IO code assuming the devs give it the thumbs up?

kidBrazil, 2018-03-15:
@totterfree awesome man no problem at all. I see no reason why we cant merge it tho I am pretty sure it is already part of the codebase. Will check.

Cheers

joeostrander, 2018-03-15:
I did my tests with Butterflight, I think it's all good :)

kidBrazil, 2018-03-20:
Closing issue as it seems to be solved by existing pinIO interface.

closed 2018-03-20 by kidBrazil