ButterFlight/butterflight #56: flashed 100 flying egg to ffpikoblx but now the controller does not see the spectrum rx the rx light is on so it has linked to the tx .I tried all the ports with no help . On configuration screen the spektrum is set up as always . if i flash back the betaflight 3.2.4 all works but if i flash betaflight 3.3.0 same problem as butterflight

garyjohnmac, 2018-03-17:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

garyjohnmac, 2018-03-18:
Update
Flashed from butterflight back to betaflight 3.3.0 same problem spektrum orange led burns solid when link to the tx  but no connection to the bourd . flashed to betaflight 2,2,5 all works normally drones flyes .I flashed back to betaflight 3.3.0 problem back and same problem on butterflight so it cant be a hardware problem as it works on betaflight 2.2.5 
Has no one a solution to my problem

kidBrazil, 2018-03-18:
@garyjohnmac - Does the port need to be inverted for spektrum?

garyjohnmac, 2018-03-18:
Dont think so dont know of they have changed the ports but the port was always on UART3 did try all the ports with no success .
 Reciever mode is set as serial based Reciever (speksat,sbus,sumb)
Reciever provider  is set at spektrum 2048 as it is dsmx sat reciever 
Channel map is set as TAER 1234

garyjohnmac, 2018-03-19:
Changed serialrx_inverted = on to set serialrx_inverted = OFF and that fixed the problem thanks all

closed 2018-03-21 by garyjohnmac