ButterFlight/butterflight #58: Can not re-arm quad after disarm from crash

dchosking, 2018-03-18:
When using a Helio spring, any small bump will cause the FC to disarm it is only possible to rearm the quad after pulling the lipo and plugging it back in

 You can reproduce this by bouncing the quad on the ground it should disarm and then you wont beable to rearm
FC: Helio Spring 
ESC: RF Bolt 4 in 1 
Checked all solder joints and RX
FW Version : BuF-3.4.3 RC3 from www.heliorc.com

Matthias7Song, 2018-03-18:
Exactly same issue here. 

FC: Helio Spring
ESC: RF Bolt 4in1 & Airbot Typhoon32
FW: Buf-3.4.3 RC3

dchosking, 2018-03-18:
I did a quick video showing how to trigure the issue in this video i did not disarm at all i dropped throttle to 'bounce' on the ground and the quad immediately disarms the same happens if you clip a race gate for example.
https://www.youtube.com/watch?v=NvBp2wcYHCM

kidBrazil, 2018-03-18:
@Matthias7Song @dchosking would either of you be willing to test a hotfix hex from @rs2k ?

dchosking, 2018-03-18:
@kidBrazil yes i am more than willing to test, i might only beable to get round to doing it tomorrow afternoon tho

kidBrazil, 2018-03-18:
@dchosking - Can you please try this hotfix hex and make sure to run imufupdate after flashing.

Thanks!

[HELIO_HOTFIX.zip](https://github.com/ButterFlight/butterflight/files/1823007/HELIO_HOTFIX.zip)

orneryd, 2018-03-18:
We believe this might be an issue that affects the older bolts and typhoons causing the F4 CPU to brown-out. I have 5 Typhoon v2.1(the latest version) and 2 Spedix GS35 4 in 1s which do not seem affected by it. but we have multiple reports of the older ones doing this. We believe it has to do with eggregious electrical noise on the vbat line from those ESCs. 

Can you try and run the flight controller voltage directly from the battery lead and see if the issue persists? If not, can you try powering it from the 5v only? (voltage in OSD will not read correctly though.)

Matthias7Song, 2018-03-19:
Tried hotfix but doesn’t fix this issue. Cannot arm after crash but can after soft landing. Bolt esc.

ArtimusPrimeFPV, 2018-03-19:
I am running th typhoon32 v2 i believe the latest version got 1 thursday from RTFQ and one Saturday from CrownCityFPV and its happened to me on both.....I am using whatever the plug wire runs...it has vbat so I figured that is how it is powering it...Am i wrong?? one from rtfq the other from crowncityfpv same is happening...brown out or wtf...
and I'm running a 1000uf 35v low esr panasonic cap on it..
but my typhoon32 v1 (wraith)  4in1 ran exactly the same way is fine....

ArtimusPrimeFPV, 2018-03-19:
Can I say though @kidBrazil that is quite a dapper picture of you...

kidBrazil, 2018-03-19:
Looks like this might be hardware related - Quoting Ornery below

> Hey all,

There have been a handful of reports (4 people that we know of) where a our Spring Flight controller might freeze when it takes a "bump" or crashes and won't re-arm until it reboots. If you are experiencing the issue, contact us directly.

We are actively investigating the cause and we believe we have it narrowed to a only a few voltage regulators which came from a faulty batch. Those handful can't handle ESC noise coming from some older high-powered 4 in 1s causing the FC to "brownout." We are also working with the manufacturer to see if we can update the design to account for natural variance found in the hobby overall. We already took certain precautions (double capacitors on each power line and an LC filter) but with such a wide variation of hardware, it's difficult account for everything.

Right now we know it might affect older Bolt and Typhoon 4 in 1s but only with the handful of faulty regulators.

Zuldan, 2018-03-19:
@kidBrazil @orneryd can you tell us which batch numbers are faulty so we can check to see if our board is affected? I purchased 2x Helio Spring FC's last week (yet to be installed). I'd prefer not to go through the hassle of wiring them in if I can find out beforehand if they are faulty.

orneryd, 2018-03-20:
@Zuldan we *believe* it is that, but we don't have 100% confirmation we are
still investigating. Right now it's literally maybe a half a dozen boards
with the issue and only when combined with really noisy ESCs. you could
also try putting a capacitor on the power line going to the FC from the ESC
and see if it resolved it. But again right now this literally only a
handful of boards out of hundreds.

orneryd, 2018-03-20:
So I have been testing tonight and I am wondering for anyone experiencing the issue, can you try powering from 5v only and see if the issue persists?

Zuldan, 2018-03-20:
Maybe if everyone could provide a little bit of information about their setup (like the OP and @Matthias7Song was doing) to help the devs. Example below ->

**Butterflight Version:** 3.3
**ESC Name:** Airbot Typhoon32
**FC Power Source:** VBAT
**BLHeli Version:** 16.7

@orneryd is there any additional information you think would be useful in diagnosing the issue?

chriswils, 2018-03-20:
You should also add the amount of external filtering, like 1000µF 24V <manufacturer/partnum> low esr cap on battery leads, 330µF on each ESC. You can also try caps for the 5V rail

orneryd, 2018-03-21:
ok guys
UPDATE: We believe we found the root cause and it is not actually the regulator. It's cable management. 

Without getting too technical, the issue is that ESC signal wires act like an antenna. If anyone does experience this issue, we suggest rerouting your esc signal wires away from any inductors or from across the board. Another possible solution is to attach another ground wire from the GRD rail on top of the board to the battery lead if you can't reroute your wires. Added capacitance will not really help in this case.

This seems to only affect 32Bit 4 in 1 ESCs that are electrically "noisy" and where the signal wires are laying across inductors on the FC.

[image]
[image]

ArtimusPrimeFPV, 2018-03-21:
My problem with that is why on my typhoon v1.1 esc also a 32bit esc built on the same platform as my other build both stingy frames, both identical builds minus the esc and the camera... Why is that fine with wires run exactly the same way?? Again I’m no electrical engineer but that doesn’t make a whole ton of sense to me when airbot claims it has more caps on the newer version and that would say to me it’s electrically quieter? So please explain that to me...because to be honest your going to have a hard time selling that to your customers who are using this for small tight racing builds or freestyle frames like the hyper low or other frames like that where you don’t have the ability to fit everything in without running the esc wiresbetween the esc and fc...

Airbot typhoon32 v2 newest one
Vbat
1000uf 35v low ear Panasonic on batt leads
Fw 32.4 blheli32

orneryd, 2018-03-21:
Things run differently at higher speeds than what was typical before. The faster things run the more ESI is generated that capacitors don’t absorb. The issue at that point is grounding. The RF signals being absorbed by components on the FC needs somewhere to go.

robert-b, 2018-03-22:
for those who have the issue.
the shutdown is way to clean for me.
it might be the receiver which causes a failsafe.
so reconfigure failsafe to land smooth (no disarm)
2nd iteration is a cap on the power line for receiver.
and those cable should be drilled as well (with the signal line).

robert-b, 2018-03-22:
move the receiver as far away as possible ... the antenna might be source.

there is a possible sw solution for that ...

ArtimusPrimeFPV, 2018-03-22:
My receiver is as far away as it can be and the antenna are out the back of the quad on a stingy frame. Also cap on the power leads doesn’t work even when I rerouted the 4-1 line around or on top or under nothing seems to fix my issue not loop times nothing...

robert-b, 2018-03-22:
ok, now reconfigure the failsafe action not to disarm.

orneryd, 2018-03-22:
Something to note, Revolt users are experiencing similar issues using the Typhoon and Bolt 32 bit ESCs/

robert-b, 2018-03-23:
one more test pls. with the no-disarm failsafe action.
I hope there is enough time during the weekend to do so.

orneryd, 2018-03-23:
Update: this only affects certain Typhoons. It's hit or miss but it seems abut half of the time the RF noise (as opposed to typical electrical noise) from a 32 bit Typhoon or 32bit variant can lock up the FC. We are taking a couple steps to try and harden the FCs but we have been able to reproduce the issue on certain Typhoons specifically.

For example, we had one Typhoon and Spring combination that would consistently lock up. We tried 3 different Spring FCs. We even tried replacing a few of the components on our FCs to see if something was at fault on our end and nothing worked. I took those same FCs and ran them on a different Typhoon V2.1 a two Spedix GS53 4 in 1s and they worked without a hitch with the original components.

robert-b, 2018-03-23:
can i get one of these not so nice fc's?
the typhoon esc should be on the way real soon.
and pls. what motors?
add a simple bluetooth dongle and log a simple data type over the wire ;-)
when it locks up you see it immediately in the trace.

gke, 2018-03-26:
Misspost reference was possible Acc Z bump detection but Butterflight appears not to use this for landing detection and motor shutdown. Have not looked at the crash detection code but that is unlikely.

Freakofspeed, 2018-03-31:
I think this is hardware related
I can report i was having Helio Spring lockup on a hard bump.  OSD would stop updating FC would not respond to switches, could not turn-on the beeper.
ButterFlight / HELIOSPRING (HESP) 3.5.0
T-motor F40 Pro  2400KV
Spedix GS32 4in1 35A ESC
Crossfire nano RX
TBS Unify PRO HV
Lumenier AXII
Foxeer Predator Mini
Panasonic LowESR 680uf 35v cap on the 4in1
Power and signal connected via 4in1 jst port.

I re-arranged my stack by changing standoff length moving the inductor away from the motor lead an haven had a re-occurrence  yet  (16 batteries so far) 

Before
[image]

After
[image]

kidBrazil, 2018-04-01:
Closing issue since it has been determined to be hardware related and affecting a small number of components. Feel free to re-open if issue resurfaces.

closed 2018-04-01 by kidBrazil

frizzle101101, 2018-05-20:
Currently having this issue with:
ButterFlight / HELIOSPRING 3.5.1
T-motor F40 Pro 2600KV
Spedix GS32 4in1 35A ESC
Low ESR cap and regular rxsr and tbs;
5 inch build.
was able to do 6 laps In a race only once shut off in air with out impacts, otherwise always happens on a crash. Going to try separating the boards a bit will report if this fixes it...can we get some insight to the electrical side of this, is it a voltage drop or noise?

 This is definitely not solved, because I have helio with GS 25a escs,
1407 3600kv
With no cap, and I have not had a brown out.
This should be solved before moving forward with the hardware because there is obviously something the fc can't take if we all have the exact same issues.

Going to try

Zuldan, 2018-05-20:
@frizzle101101 this is problem a question for the Helio Slack channel "flyhelio".

You need to flip your Helio board upside down to avoid the FC lock up. The problem is caused by the EMI noise from the 4in1. 

4in1 manufactures are now releasing their 4in1's with some copper tape.

Que60, 2019-08-22:
Did anyone ever find a so;oution to this issue? I ave tried all the suggested fixes with no luck.