ButterFlight/butterflight #59: Cli issues!

jwgreening, 2018-03-18:
I flashed the new 3.4.2 and 3.4.3 to my helio spring.  Went to enter the commands on the butterflight site into the cli, but nothing is recognized. HAs there been a change to the firmware? No kalman filter, or overclocking. Or is this a helio spring issue.

jwgreening, 2018-03-18:
Here is the diff file.
# diff

# version
# ButterFlight / HELIOSPRING (HESP) 3.4.2 Mar  9 2018 / 18:27:23 (5ef2a93) MSP API: 1.38

# name
name JAYBIRD FPV

# resources

# feature
feature -AIRMODE
feature DYNAMIC_FILTER

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map
map EATR1234

# serial

# led

# color

# mode_color

# aux
aux 0 0 3 1900 2100 0
aux 1 1 1 1900 2100 0
aux 2 2 1 1400 1600 0
aux 3 13 5 1900 2100 0
aux 4 26 4 1900 2100 0
aux 5 28 1 900 1100 0
aux 6 35 5 1400 1600 0

# adjrange
adjrange 0 0 2 900 1150 12 2
adjrange 1 1 2 1375 1625 12 2
adjrange 2 2 2 1850 2100 12 2

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 1
set rssi_channel = 16
set serialrx_provider = CRSF
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set failsafe_kill_switch = ON
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 2
set osd_warnings = 63
set osd_tim1 = 2561
set osd_vbat_pos = 2052
set osd_rssi_pos = 2061
set osd_tim_1_pos = 2067
set osd_vtx_channel_pos = 2435
set osd_current_pos = 2454
set osd_craft_name_pos = 2442
set osd_warnings_pos = 2121
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 60

# rateprofile
rateprofile 0

set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 10
set roll_srate = 68
set pitch_srate = 68
set yaw_srate = 68

kidBrazil, 2018-03-18:
@jwgreening - Those settings are for NON HELIO boards only. HELIO has its own filtering separate from the firmware so FKF and all the other filters are not needed. The only thing you need to worry about with your HELIO is PID.

closed 2018-03-18 by kidBrazil