ButterFlight/butterflight #60: YAW Spin to the Moon

frakkan, 2018-03-19:
Death of yaw spin. Please re-add the feature, to prevent yaw spin to the moon. After clipping a gate, or a tree, the gyro goes nuts, and the quad spins to the left, causing either a crash, or travel to the moon. 
Betaflight had this issue, they fixed it, now please put the code back to butterflight. It makes using the fw very dangerous.

kidBrazil, 2018-03-19:
@frakkan We are still debating what we are going to do regarding that feature. Will get back to you with an answer when we decide.

chriswils, 2018-03-20:
What are the talking points of that debate? Is it unsafe to enable them ourselves from the CLI?

frakkan, 2018-03-20:
The point is,  to find out exactly what we need to enable in cli.  Not
everyone a firmware programmer..

chriswils, 2018-03-20:
@frakkan I was pointing to kidBrazils [comment](https://github.com/ButterFlight/butterflight/issues/60#issuecomment-374260403), not your issue, when asking for what this debate involves.
If you are looking for the CLI command, look at  `gyro_overflow_protection`.

kidBrazil, 2018-03-20:
We talked about it and decided to turn it back on to YAW as a default - it can still be changed via CLI command with gyro_overflow_protection. Should be resolved with #62

@frakkan Fair but this is experimental software, so a little tinkering is to be expected =)

gke, 2018-03-23:
I provide the original solution to the BF YTTM. My background is control engineering amongst other stuff.

The specific problem is akin to two's complement overflow in the ICM gyro ADC when the rate exceeds the maximum value. The fix is to detect when the rate is high and the direction of the rate changes suddenly between readings and to hold the old value until the yaw PID drags the rate back.  

The fix is marked by the USE_GYRO_SLEW_LIMITER define in gyro.c.

Unfortunately the BF dev guys  complicated it later and just generally resisted its inclusion but it is still there in Butter. Yaw is the most likely to spin up and go through 2000 deg/s  but with the smaller quads it may still be possible to spin through 2000 deg/s.  So as the code has minimal execution time I would apply it to all axes and not bother making it conditional on it being for the bad ICM being used.

The BF dev guys also added code to turn off yaw control as well which shows a failure to understand Control 101.  That code that turns of the yaw PID should definitely go as should the code in pid.c wherever "inCrashRecoveryMode" occurs as the Level mode in BF is poor in accelerated fight. As a consequence activating Level mode in desperation when the angle estimates are so poor is questionable - check the artificial horizon in flight videos.  The pid.c code needs a bit of work to clean it up generally.

I hope this helps as Butter seems a little more friendly.

robert-b, 2018-03-23:
hi greg,
i modded the stuff and tamed the pid calulations (d-term).
flies nicely on an spracingf4evo with the icm20608.
your slewratefilter in place ;-)
i start slowly with some simple changes in front.

pid.c is a wow moment and the wtf count is pretty high.

kidBrazil, 2018-03-23:
@gke Thank you very much for the input Greg - I will look into the USE_GYRO_SLEW_LIMITER feature you mentioned and discuss re-implementing it as the default.

I dont think that will be a problem with the other devs at all. We really appreciate your collaboration!

gke, 2018-03-24:
@robert-b  Hi Robert. Glad to see you here as well.  Yes there is a lot of wtf stuff that needs to go into the bit bucket.  

@kidBrazil  Thanks Lucas.  Just now looking at the current master in gyroUpdateSensor you have applied the filter as I originally proposed it i.e.  to all axes.  

Feel free to point me at stuff I may be able to help with.  Hopefully I can make a small contribution. 

Cheers

kidBrazil, 2018-03-24:
@gke That's a very kind offer on your part and I will keep that in mind as we go through all of this stuff here. We would love to have as many capable contributors as possible. If I find something I will definitely let you know. In the mean time, feel free to poke around and submit PR's any time. =D

xermalk, 2018-05-15:
Betaflight 3.4 will by default have a new feature called "Yaw Spin Recovery" enabled.
It fully disables the pid controller and uses all available motor power until a yaw spin has been tamed.

I tested it for a while last week, and its a game changer. As long as you don't loose props you will never experience another yaw spin again. 

https://github.com/ctzsnooze/betaflight/wiki/Draft-wiki-notes

Is this something that will be ported to Butterflight?