ButterFlight/butterflight #71: Systemconfig

robert-b, 2018-03-24:
pid loop frequency is not settable for mpu-6000 fc's.
default setup always remains after booting.

works with betaflight but not with butterflight.

robert-b, 2018-03-25:
i bought a realacc set with all components.
that set did include an omnibusf4sd.
which is no used for desktop flying.

i did load the defaults together with a binary from github making sure my changes do not interfere.
for me it fails using the 10.2.0 configurator.

robert-b, 2018-03-25:
now i see clearer.
8/2 is max possible setting allowed for the mpu-6000.
what if you have an f7 - 8/8 is possible without question.

some sort of teaching what to use?

ABLomas, 2018-03-25:
8K is max gyro rate on MPU6K, so 8/8 is possible (also something weird like 8/16 is possible on F4, but no sense to run pid loop faster than gyro readout).
8/8K runs on F4 omnibus without issues. Which exactly pid loop frequency is not settable?

robert-b, 2018-03-25:
i can only select 8/2 as the max.
8/1 is possible - 8/8 not.
regardless if f4 or f7.

which version of the config tool do you run?

ABLomas, 2018-03-25:
10.2.0
Maybe you run too many features and CPU could not keep with it, etc?
Try simple test - do "defaults" and just change looptime to 8/8. Should work.

robert-b, 2018-03-25:
you explained what i did.

ABLomas, 2018-03-25:
I think i know where your issue is.
I took Omnibus F4 board, flashed with F4SD FW (no difference between boards, only output mappings).
And yes, on defaults you cannot set PID loop higher than 2K, because ESC algo is set to "synced" by default and is set to ONESHOT125.
So, if you want to run synced - you can't use oneshot at higher than 2K speeds.
Solutions:

1. use DSHOT - and you can select 8/8
2. turn off synced ESC output mode - and you can select 8/8

IIRC this is even documented somewhere - def. not configurator/firmware issue.

robert-b, 2018-03-25:
good point.
i will close the issue as it works.

closed 2018-03-25 by robert-b

robert-b, 2018-03-25:
https://github.com/robert-b/butterfly
has some changes for the gyro and pid stuff ...
the new slewratefilter is active for all gyros.
for those hitting gates during a race ...