ButterFlight/butterflight #73: pid setpoint calculation is wrong

robert-b, 2018-03-24:
`    dtermSetpointWeight = pidProfile->dtermSetpointWeight / 127.0f;`

when using full, 100 % , dterm weighting weighting it is in the configurator displayed as 1.27
you should team up getting the calculus right ...

kidBrazil, 2018-04-01:
@robert-b I am a bit confused here, you are saying that when you set Dterm setpoint weight to max value in configurator it gets displayed as 1.27? Where does that get displayed? UI?

robert-b, 2018-04-01:
it is the slider in the pid tab.
setpoint weighting would be the english term.

orneryd, 2018-04-15:
I think it's a display "feature" just like the pids and other values which are displayed differently and then scaled when used.

closed 2018-04-15 by orneryd