ButterFlight/butterflight #75: CPU locks up shortly after arming?

elitak, 2018-03-25:
Shortly after arming, or immediately if I throttle up, the OSD locks up and the motors stop. I have no props on, am running 6s with no current sensor, on HELIOSPRING v3.4.2, 32/16kHz. X-racer 32bit Quadrant ESCs running BLHeli v32.2. It feels like I'm missing something basic, but I've never used this FC or ESC set before, so I can't really tell. I also can't fall back to Betaflight for this board to see if the problem is in the Butterflight firmware.

Elided sections are defaults:
```
# diff
 
# version
# ButterFlight / HELIOSPRING (HESP) 3.4.2 Mar  7 2018 / 18:18:45 (2dc6f02a5) MSP API: 1.38
 
# feature
feature DYNAMIC_FILTER

 
# map
map TAER1234
 
# serial
serial 1 8192 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 4 1 57600 57600 0 115200
 
# aux
aux 0 0 0 1300 2100 0
aux 1 1 1 1700 2100 0
aux 2 13 2 1300 2100 0
aux 3 31 3 1700 2100 0
aux 4 35 2 1700 2100 0
 
# master
set gyro_sync_denom = 1
set rssi_channel = 12
set fpv_mix_degrees = 30
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = 180
set current_meter = NONE
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set osd_warnings = 63
set osd_rssi_pos = 2119
set osd_current_pos = 195
set osd_mah_drawn_pos = 163
set osd_craft_name_pos = 2432
set osd_avg_cell_voltage_pos = 2124
set vtx_band = 5
set vtx_freq = 5658
set vcd_video_system = 2
```

gke, 2018-03-25:
Further to Roberts comments it could be a false detection of an AccZ bump which is used to detect the impact on auto landing. I will try to have a look.

Rant: Just generally however "modern" board design seems to think capacitors are not needed. Take the circuit design as provided by an engineer (if he/she exists) and keep taking components off the design until it stops working then put the last one back. Capacitors are the first to go as they are more expensive.

gke, 2018-03-26:
Thanks Eric .

orneryd, 2018-03-28:
yeah see #58 for more information. We really need to disabled dynamic in the UI when it's the helio target. It doesn't do anything since the code is removed :)

elitak, 2018-03-31:
This is a dupe, I guess, but my failure was almost immediate, in contrast. I fixed the problem by spacing the board from the ESCs a good distance (needed about 10mm I think?). The inductor on the FC was only a millimeter or so from a row of capacitors on the ESC, so probably some EF was causing the brownout.

closed 2018-03-31 by elitak