ButterFlight/butterflight #80: Please create a cli command to turn on the disarm switch that was removed

WiKDMoNKY, 2018-03-27:
BF 3.3.0 disabled the ability to disarm in the air, which in my opinion, is dangerous for all of us that have been landing using this setting since CleanFlight. Why can't there be a cli command that turns this back on?

kidBrazil, 2018-03-28:
It goes against RC convention to have the arm switch not be an actual kill switch. I know you guys have been doing this for years but I am sorry to say it is an unsafe practice.

We are not opposed to including it as a CLI option, provided the default of that option is for it to be OFF. Meaning, when the arm switch is killed, the quad drops. Because that is expected behavior and safer behavior.

That said, we have already discussed this point, none of the Butter devs have time or interest in this feature at this moment. You are more than welcome to submit a PR with a solution  and we will review it and discuss it fairly.

Cheers

WiKDMoNKY, 2018-03-28:
You will see a lot more bitching about this feature being removed as us old timers begin to upgrade to BF 3.3x and BuF 3.x.x. Count on it ;)

kidBrazil, 2018-03-28:
@WiKDMoNKY That is fine - maybe one of them has some coding chops and they can just submit a PR for us to review =D

WiKDMoNKY, 2018-03-28:
I doubt it, but there are a couple of us working on a modified BuF firmware that will put the setting back in. We may possibly make our own fork. Sometimes ya gotta do what you gotta do when Big Brother tries to tell seasoned pilots what is safe for us ;)

therealjonnyd, 2018-03-29:
I keep seeing the same people (person) bringing this up. I don't think it's the widespread issue that it's being made out to be. It's inherently unsafe, therefore it was removed. In all fairness, it shouldn't have EVER been an option. By all means, set it up in your radio at your own risk if you'd like but this reminds me of my grandfather cutting the seatbelts out of his Buick because he didn't believe in them.

I highly suggest you relearn how to land. It's very simple to land with airmode off and disarm completely upon touching down.

theSkonk, 2018-04-05:
or just do what most racers do... get low, slow and kill switch the beast!

iAMsinister, 2018-04-05:
how does that make sense? "I doubt it, but there are a couple of us working on a modified BuF firmware that will put the setting back in. We may possibly make our own fork."

why not just submit a PR like Loose asked? landing is honestly not that hard even with airmode on.. ill gladly post a video demonstrating me landing on a 30m high dive with airmode on disarming and taking off again, on a windy af day mind you. i kno the whole saying "old dog new tricks.." but thing is y'all where practicing an unsafe landing procedure from the get go..

WiKDMoNKY, 2018-04-05:
I/we are over trying to get this changed and are dealing with learning to land "the safe way" that big brother deems as necessary to be safe.

frakkan, 2018-04-05:
Did I miss something?  Where was the disarm switch removed?  I am flying
3.5fw and have an arm switch on an aux channel.  Anytime i can disarm my
quad.... Sorry for asking,  but just want to understand the debate.

theSkonk, 2018-04-06:
@frakkan in older beta flight versions you could disarm while flying  and the 'disarm' would not come into effect until you actually shut the throttle..