ButterFlight/butterflight #86: Fix: 3D flymode improvement

thongraegu, 2018-03-30:
https://github.com/betaflight/betaflight/pull/5571

That PR Iw been testing, makes 3D mode fly just like it should, been testing it for a while. It doesnt affect/change anything on normal 2D side of things, just effects 3D. :)

Teracis, 2018-03-30:
The PR referenced is just for testing via disabling the mixer gain halving using the cli variable. If it's found that the halving should be removed permanently the PR will need to be updated and merged into the master. Testing is still in progress, thanks for helping with the testing @thongraegu

thongraegu, 2018-03-31:
Alright, this is the real deal fix

https://github.com/betaflight/betaflight/pull/5582

varkokonyi, 2018-03-31:
Cherry-picked this commit over from BF, just for you @thongraegu

Teracis, 2018-04-01:
It should help everyone who wants to fly 3D. I meant to get back here and reference it but glad you guys found it. I got to test it myself today and it's epic!

varkokonyi, 2018-04-01:
I have seen the discussion on the BF repo and the issue and thought the BuF guys could use this

PR #90 (2018-04-01): Removed mixer halving in 3D mode.
Fixes #86.

Cherry-picked from betaflight#5582
closed 2018-04-01 by kidBrazil