ButterFlight/butterflight #87: BuF 3.5.0 ESC issue

HotwiresFPV, 2018-03-31:
updated one of my quad Fury F3S FC with Spedix GS35A 4in1 ESC from BuF 3.4.2 to BuF 3.5.0 and now only motors 2 and 4 are spinning. plugged it to BLHeli 32 Configurator and it only see ESC 2 and 4. revert back to BuF 3.4.2 all motors spinning and all ESC are present on BLHeli 32 Configurator. I have tried multiple ESC firmware available and ended up with same results

Zuldan, 2018-03-31:
are you running proshot or dshot? and is your ESC firmware version?

HotwiresFPV, 2018-03-31:
Either. Proshot, Dshot or Multishot

HotwiresFPV, 2018-03-31:
Esc firmware flashed 32.1, 32.3 and 32.4. All of these firmware work on BuF 3.4.2

Zuldan, 2018-03-31:
Try DSHOT600 or DSHOT1200 on 3.5.0. I had motors not starting up when using PROSHOT1000.

HotwiresFPV, 2018-03-31:
I did try DShot1200 and DShot600. Even Multishot and have the same results.

orneryd, 2018-04-04:
have you tried setting dshot_burst=ON?

HotwiresFPV, 2018-04-04:
Dshot burst is on

orneryd, 2018-04-15:
is this still an issue in 3.5.1?

kidBrazil, 2018-04-15:
@orneryd Looks like it is yes

HotwiresFPV, 2018-04-17:
Well... I cannot flash to 3.5.1 because it breaks F3. So to me issue is not fixed

kidBrazil, 2018-04-19:
Can you folks who are having this issue please try the hexes contained here - https://www.dropbox.com/sh/111yluuy3aendin/AABwxKeUC4H2zDBMYfA-2kUca?dl=0

Let me know if you are able to arm and it all works as it should.

Cheers

LexioTech, 2018-04-20:
@dmagicone Can you please post your CLI dump here? It would help us look deeper into this.
Thanks!

dmagicone, 2018-04-22:
Sure, glad to help.
[BTTR_cli_20180421_201446.txt](https://github.com/ButterFlight/butterflight/files/1935118/BTTR_cli_20180421_201446.txt)

HotwiresFPV, 2018-04-23:
Issue was not solved with the posted test firmware only motor 3 and 4 are spinning. However it works fine with 3.4.2 firmware.

kidBrazil, 2018-04-25:
One more set of hexes for you folks to try plz - https://www.dropbox.com/sh/111yluuy3aendin/AABwxKeUC4H2zDBMYfA-2kUca?dl=0

This one removes LED support again.

dmagicone, 2018-04-27:
No spracing f3 hex to try.

dmagicone, 2018-04-27:
You think it could have something to do with how after 3.4.2 the board alinement is different? On all version before the alinement I used was 270. In 3.5.0 and newer I had to use -180. If I used 270 like always, the quad would move the wrong direction in the configurator.