ButterFlight/butterflight #91: 3.5.0 DShot protocol problem

kokein, 2018-03-31:
After upgrading and flashing up to BuF 3.5.0 now only motor #1 is spinning. 
From time to time motor 3 will "bulb/shake" a little.
Done full chip erase while flashing via GUI, always do. 

hglrc f3 v4 pro (spracingF3)

It was all ok on 3.4.2 version 

```
# dump
# ButterFlight / SPRACINGF3 (SRF3) 3.5.0 Mar 29 2018 / 14:41:33 (dd56ee0) MSP API: 1.39
name GTR2 =KKN=
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource SONAR_TRIGGER 1 B00
resource SONAR_ECHO 1 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource LED 1 B03
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 NONE
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 NONE
[... 497 lines cut ...]
set i_yaw = 45
set d_yaw = 20
set p_level = 50
set i_level = 50
set d_level = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 2

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 192
set pitch_rc_rate = 192
set yaw_rc_rate = 192
set roll_expo = 22
set pitch_expo = 22
set yaw_expo = 12
set roll_srate = 68
set pitch_srate = 68
set yaw_srate = 59
set tpa_rate = 10
set tpa_breakpoint = 1650
```

SpeedMetal1, 2018-04-03:
Same here.  XRacerF3.

orneryd, 2018-04-04:
did you try:
```set dshot_burst=ON```

kidBrazil, 2018-04-04:
@SpeedMetal1 @kokein Can you please try the dshot_burst=ON and report back? if that works I know what the fix is.

SpeedMetal1, 2018-04-05:
Sorry for the delay. I tried it only motor 1 spins. This is with Dshot600.

johnblackj, 2018-04-06:
Same on my Kakute V1 spracing F3, 3.5.0 only motor 1 spins. on 3.4.2 everything works fine

kokein, 2018-04-08:
@kidBrazil Hi, nope. Same thing. Tested all dshot's, only Multishot is ok.

kidBrazil, 2018-04-08:
@SpeedMetal1 @johnblackj @kokein - Thanks guys, I will keep digging here to see what happend.

Ety-T, 2018-04-15:
Same on SP Racing F4 EVO. All OK in Betaflight and ButterFlight 3.4.2

dronejunkie, 2018-04-16:
Mine is omnibusf4sd, I could not get my quad arm at all. I had to got back to 3.4.2. I am using dshot600.  I checked receiver tab, receiver is working. Checked mini_check and CPU load are under 10%. I do not have acc and baro enabled.  I could spin up motors inside motor tab.

dronejunkie, 2018-04-16:
Mine is omnibusf4sd, I could not get my quad arm at all. I had to go back to 3.4.2. I am using dshot600.  I checked receiver tab, receiver is working. Checked mini_check and CPU load are under 10%. I do not have acc and baro enabled.  I could spin up motors inside motor tab.

kidBrazil, 2018-04-19:
Can you folks who are having this issue please try the hexes contained here - https://www.dropbox.com/sh/111yluuy3aendin/AABwxKeUC4H2zDBMYfA-2kUca?dl=0

Let me know if you are able to arm and it all works as it should.

Cheers

SpeedMetal1, 2018-04-20:
Arms just fine. But the smart audio is still not working. Betaflightf7 / AKK Mach 2.

What does not work is the vtx low power disarm. I have my vtx set to 25 but when armed it switches to 800. That part does not work. 

Thanks for your help.

Ety-T, 2018-04-21:
SP Racing F4 EVO target, motors started to work in Motor tab. But motors changed their order. Now they are 4,2,3,1 instead of 1,2,3,4. I now it's not a big deal to change motors order using remapping but for information.

SpeedMetal1, 2018-04-21:
> 
> @dronejunkie <https://github.com/dronejunkie> @Ety-T <https://github.com/Ety-T> Thanks for the info so far its very helpful.
> 
> Could you both please post dumps? If you add them as text files extra points =D

dump

# version
# ButterFlight / FPVM_BETAFLIGHTF7 (FBF7) 3.5.1 Apr 17 2018 / 13:28:58 (2363d47) MSP API: 1.40

# name
name -

# resources
resource BEEPER 1 D15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 E00
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
[... 587 lines cut ...]
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 80
set pitch_expo = 80
set yaw_expo = 40
set roll_srate = 83
set pitch_srate = 70
set yaw_srate = 65
set tpa_rate = 10
set tpa_breakpoint = 1650

#

Dirtbrother, 2018-04-21:
Having a very similar issue with my BabyHawk. Was on Betaflight with DShot 600 working and came to Butter to try. 

Flashed  butterflight_3.5.0_SPRACINGF3EVO.hex and got it all working but no control over motor 4 in DShot. 
Switched back to Multishot and it works fine.

Flashed the hex posted earlier in this thread as a fix and no difference.

CLI dump attached from patched hex.

Cheers!
[BTTR_cli_20180421_201543.txt](https://github.com/ButterFlight/butterflight/files/1934917/BTTR_cli_20180421_201543.txt)

dronejunkie, 2018-04-22:
Please find attached files.  One is diff and one is the full dump

diff

# version
# ButterFlight / OMNIBUSF4SD (OBSD) 3.4.2 Mar  7 2018 / 18:10:12 (2dc6f02a5) MSP API: 1.38

# name
name SX5

# resources

# mixer

# feature
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_filter_q = 1600
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set serialrx_provider = SBUS
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT600
set failsafe_kill_switch = ON
set pid_process_denom = 1
set osd_vbat_pos = 2497
set osd_rssi_pos = 2093
set osd_tim_1_pos = 2103
set osd_throttle_pos = 2456
set osd_current_pos = 2518
set osd_craft_name_pos = 2082
set osd_warnings_pos = 2505
set debug_mode = GYRO

# profile
profile 0

set dterm_notch_hz = 0
set anti_gravity_gain = 3000
set setpoint_relax_ratio = 20
set dterm_setpoint_weight = 100
[... 674 lines cut ...]
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 51
set pitch_srate = 45
set yaw_srate = 73
set tpa_rate = 10
set tpa_breakpoint = 1650

#

kidBrazil, 2018-04-25:
One more set of hexes for you folks to try plz - https://www.dropbox.com/sh/111yluuy3aendin/AABwxKeUC4H2zDBMYfA-2kUca?dl=0

This one removes LED support again.

dronejunkie, 2018-04-27:
Still no good.  Can not arm.

SpeedMetal1, 2018-04-27:
BetaflightF7 arms and flies good.  Butterflight Lua scripts do not work.  Also, tried RC2 with XRacerSPI, only motor one spins with DShot600.  This quad also has a Mach2 vtx, which works, along with the Butterflight Lua scripts.

dronejunkie, 2018-04-27:
@kidbrazil, is there any debug I can turn on which could help to debug this issue?

Mitchlol, 2018-05-07:
I'm having this issue too, motors 1 & 2 work fine, 3 and 4 don't spin, i'm uSing the bardwell f4 board (Target: CLR4)

As mentioned above, this only happens on d-shot, works fine with multishot.

Jackalneck, 2018-05-09:
Same problem on CLRacingF4.  Only two motors will arm if LED feature is enabled.  Disabling LED strip fixes problem.

iespoto, 2018-07-08:
Same here.
Using BuF with SPRACINGF3v1 + RS20A V2 ESC's.
Same thing - only motor #1 is spinning.
Changing between protocols cause ESC to restart in motor tab.
Works perfect with MULTISHOT.

Ub6ib9, 2018-07-08:
I had the same problem I had to roll back firmware to keep D shot.

Kyle

iespoto, 2018-07-08:
Can confirm, that downgrading 3.4.x solved issue, and DSHOT works flawlessly.

ghden, 2018-07-25:
I have the same problem with Hypetrain Rotor Riot FC this runs OmnibusF4 firmware.

dronejunkie, 2018-07-25:
At the moment this issue is only acknowledged in the last release change
log as a f3 issue. I have seen enough reports from f4 owners and me
included it is an issue on f4 also.

Looking at the GitHub pulse for this project, there has not been much
activities in the last few months sadly.

J

I have the same problem with Hypetrain Rotor Riot FC this runs OmnibusF4
firmware.

tobiasbischoff, 2018-12-17:
What up with this? I still cant use my Kis32A ESCs with the latest Release.. Intermitted DSHOT Connectivity. Is there any fix?

rephlex, 2019-03-13:
Same problem here on SPRACINGF3 in HGLRC SpRacing F3 V4 AIO. Only one motor running in dshot, another motor twitches, two others are dead. Multishot works flawless. 3.5.4 and 3.5.6 are tested and affected.