ButterFlight/butterflight #110: Acc calibration

adrianmiriuta, 2018-04-12:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

orneryd, 2018-04-12:
Calibration doesn't "level" the quad. it levels it relative to gravity. If a table is 1 degree off it will be 1 degree off. 
[image]

[image]

closed 2018-04-12 by orneryd

adrianmiriuta, 2018-04-13:
@orneryd 
If the FC is mounted with a few degrees misalignment (2° roll 3°pitch),
or the MPU is soldered with a small angular offset ... 
You have to correct this offsets trough acc Calibration.
(put quad level and calibrate !)
Else you will get a offset in Attitude , and a drift in Level and Horizon modes !

orneryd, 2018-04-13:
calibrating it won't make it flat in the UI, calibrating it will set the
trims. if you go to the cli and type "set acc" you will see the trims set
to compensate for the angle.

adrianmiriuta, 2018-04-13:
@orneryd 
I see ,
but it should level the Attitude quaternion,

It only fails to do so because someone deactivated acc handling in 
`imuCalculateEstimatedAttitude`
`bool useAcc = false;`

This means Attitude is only calculated from gyro and will drift !

I tried to patch that (PR #112), but it was reverted !

orneryd, 2018-04-13:
I fixed that in the refactor i did here:     

```quaternion vError = VECTOR_INITIALIZE;
    quaternion vGyroAverage;
    gyroGetAverage(&vGyroAverage);

    quaternion vAccAverage;
    if (accGetAverage(&vAccAverage) && accIsHealthy(&vAccAverage)) {
        applyAccError(&vAccAverage, &vError);
    }
```
However, that code does not get used with imu-f because we calculate the quaternion and pass it directly.