ButterFlight/butterflight #115: RC7-Caprica HELIOSPRING BBL does not show debug traces

adrianmiriuta, 2018-04-13:
I tried
debug_mode=NOTCH
debug_mode=ANGLERATE
debug_mode=IMU
none works.

[image]

orneryd, 2018-04-13:
There is no Notch so we should probably remove it from our target

IMU we could probably add logging the attitiude/quaternion values coming from IMUF.

adrianmiriuta, 2018-04-14:
@orneryd 
I also tried debug_mode=ANGLERATE
it does not BBL , or MSP log either.

adrianmiriuta, 2018-04-15:
@orneryd 
@rs2k 

correction
debug_mode=ANGLERATE works (my receiver was broken)

other debug_modes are not relevant on HELIOSPRING

orneryd, 2018-04-15:
I think we are going to leave the additional modes there for potential future use, but just add some documentation stating they do not apply.

closed 2018-04-15 by orneryd