ButterFlight/butterflight #123: Butterflight 3.5.1 RC (taz)

adrianmiriuta, 2018-04-16:
tasmanian devil ...

build working with 3.5.0 flawless
configured with .diff  paste
on 3.5.1 RC 
without any stick input it looks like 

[image]

adrianmiriuta, 2018-04-16:
@therealjonnyd 
sure I triple checked !
it isn't a new setup
I only updated to  3.5.1 RC

i flashed back to 3.5.0 and it flies like a charm

```
# diff

# version
# ButterFlight / OMNIBUSF4SD (OBSD) 3.5.1 Apr 16 2018 / 08:32:52 (632dbc7a0) MSP API: 1.40

# name
name warp-270

# resources

# mixer

# feature
feature -OSD
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map
map RETA1234

# serial
serial 0 1024 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1950 2075 0
aux 1 1 0 1525 1650 0
aux 2 2 0 1700 1825 0
aux 3 6 2 1950 2075 0
aux 4 7 3 1950 2075 0
aux 5 13 1 1425 1550 0
aux 6 26 1 1950 2075 0
aux 7 31 2 925 1050 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_stage2_filter_type = NONE
set gyro_offset_yaw = 97
set gyro_overflow_detect = OFF
set acc_calibration = -15,-185,-13
set mag_hardware = NONE
set min_check = 1000
set max_check = 2000
set rc_interp_ch = RPYT
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 90
set ibata_scale = 240
set small_angle = 180
set deadband = 7
set yaw_deadband = 11
set pid_process_denom = 1
set tlm_halfduplex = OFF
set frsky_unit = METRIC
set debug_mode = ANGLERATE

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 160
set dterm_notch_cutoff = 110
set p_pitch = 61
set i_pitch = 37
set d_pitch = 31
set p_roll = 61
set i_roll = 37
set d_roll = 31
set p_yaw = 61
set i_yaw = 37
set d_yaw = 31
set d_level = 50
set horizon_tilt_effect = 50
set horizon_tilt_expert_mode = ON

# rateprofile
rateprofile 0

set thr_mid = 14
set thr_expo = 67
set tpa_rate = 70

#
```

[LOG00001.BFL.zip](https://github.com/ButterFlight/butterflight/files/1916014/LOG00001.BFL.zip)

adrianmiriuta, 2018-04-16:
@kidBrazil 
@orneryd 

and CPU crashes  on 

> status 

command !

adrianmiriuta, 2018-04-18:
@kidBrazil 
@orneryd 
update ...
the CPU crash is ralated to issue #126
and gets fixed with pr #129

the "taz" is an other issue
related to pid oscillation buildup 
probably due to changes in pid.c but I could not locate the error yet.

adrianmiriuta, 2018-04-19:
@kidBrazil 
@orneryd 

```
set align_board_yaw = 0
set align_gyro = CW0
set align_acc = CW0
```

fixes the problem !
was there a PR that changed board orientation,
or default sensor orientation ?

kidBrazil, 2018-04-21:
Closing issue since the FIX andrian provided has been merged

closed 2018-04-21 by kidBrazil