ButterFlight/butterflight #128: Butterflight 3.5.1 RC2 legacy AKK / Mach2 not working.

SpeedMetal1, 2018-04-17:
Doesn't seem to be working anymore.  I have tried both LEGACY_SA_SUPPORT on/off in the config tab, but no difference.  Nothing else has been changed.  Is there a custom firmware I need to use like in Betaflight?  

'''# version
# ButterFlight / FPVM_BETAFLIGHTF7 (FBF7) 3.5.1 Apr 17 2018 / 13:28:58 (2363d47) MSP API: 1.40

# name

# resources

# mixer

# feature
feature TELEMETRY
feature RSSI_ADC
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
feature LEGACY_SA_SUPPORT

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map

# serial
serial 1 32 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1300 0
aux 1 1 1 1700 2100 0
aux 2 2 1 1300 1700 0
aux 3 13 2 1700 2100 0
aux 4 26 3 900 1300 0
aux 5 35 3 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_to_use = 1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set dshot_idle_value = 200
set motor_pwm_protocol = DSHOT1200
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warnings = 63
set osd_vbat_pos = 2432
set osd_rssi_pos = 2075
set osd_current_pos = 2455
set vtx_band = 5
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = 2

# profile
profile 0

set dterm_notch_hz = 0
set vbat_pid_gain = ON
set crash_recovery = ON
set horizon_tilt_expert_mode = ON

# rateprofile
rateprofile 0

set roll_expo = 80
set pitch_expo = 80
set yaw_expo = 40
set roll_srate = 80'''

LexioTech, 2018-04-17:
Hey @SpeedMetal1 ,
- Did you upgrade from 3.5.0 ? Did it work then? If not, what was the last version that worked for you?
- What VTX do you exactly have?
- Did you copy your diff over from your previous config, or did you re-do everything 1by1?

Thanks, 
Lexio.

SpeedMetal1, 2018-04-17:
Hey Lexio,

I did upgrade from 3.5.0.   It worked in 3.5.0.
I have a Mach 2.
I re-did everything from scratch.  

Thanks for your help,
Scott

redneonrt, 2018-04-21:
I'm having the same problem, I upgraded from 3.5.1 RC1 to RC2 and then it stopped working.  I didn't copy anything over, reconfigured from scratch.

CLI Dump
[BTTR_cli_Rooster5_20180421_123455.txt](https://github.com/ButterFlight/butterflight/files/1934809/BTTR_cli_Rooster5_20180421_123455.txt)

This is the vtx I have 
https://www.amazon.com/AKK-Transmitter-Support-Configuring-Betaflight/dp/B075ZQ53ZX/ref=sr_1_3?ie=UTF8&qid=1524332036&sr=8-3&keywords=akk+vtx&dpID=51awSzovmvL&preST=_SY300_QL70_&dpSrc=srch

SpeedMetal1, 2018-04-21:
I think I found what part of the problem is.  The UART's have changed.  My smart audio used to be on UART3, but now on this version of Butterflight it's on UART1.  SmartPort has moved from UART2 to UART3.  redneornrt try swapping them, might help.

kidBrazil, 2018-04-25:
@LexioTech What do you think?

SpeedMetal1, 2018-04-26:
Super busy (wife and I have 4 month old triplet girls) but I installed it and it flies fine. The only thing now is my lua scripts aren’t working. I can live without that. I will check back in tomorrow when I’ve had more of a chance to fly. 

Thanks for all your hard work. 
Scott

closed 2018-06-07 by orneryd