ButterFlight/butterflight #131: Lumenier Alpha AIO flips on take off with same code as rc1

lizardguns, 2018-04-18:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

AlexSamayoa, 2018-04-19:
If you board is offset IE mine is -90 (turn to left counter clock wise pig tail on the right of craft ) now appears on RC2 as 90... i have reported this also.

kidBrazil, 2018-04-20:
Closing this as duplicate #132 and #123

closed 2018-04-20 by kidBrazil