ButterFlight/butterflight #133: 3.5.1 RC2 ESC telemetry current signal flickering/not counting mAh correctly

oyvinla, 2018-04-20:
Omnibus f4 / xracer, blheli32 quadrants.

The current sensor signals from the ESCs is not picked up correctly by the FC. The value is coming and going and does not count correctly. 8k/8k, DShot1200

JesseDFBlue, 2018-04-30:
Same here on Helio spring, also the voltage drops to 0V when esc sensor is selected for voltage monitoring.

therealjonnyd, 2018-04-30:
This is a known issue in blheli_32 when using DSHOT1200, it also happens on Betaflight. Try DSHOT600 or PROSHOT1000.

IcyLemur276, 2018-05-03:
Out of the box even on multishot the amp readings are WAY off as in it shows I'm pulling 80+amps with the motors off and just plugged in.  This is on the Helio Spring FC

therealjonnyd, 2018-05-03:
ESC telemetry requires DSHOT/PROSHOT to work.

IcyLemur276, 2018-05-04:
@therealjonnyd I'm using the DALRC Engine 40amp 4in1 ESC.  I bridged the CRNT pads on my Helio Spring FC.  Voltage readings are fine, but AMP readings are way off and don't work.  For example as i'm flying it shows 80+amps no matter what my stick values are.  Attached is my wiring diagram.
[image]

IcyLemur276, 2018-05-04:
@therealjonnyd  Current source is set to onboard ADC and i'm running 3.5.1 RC2 IMU-F V 106 and Configurator version 10.4.2.

macataxis, 2018-05-09:
Same issue here. Helio + HW 4in1 Dshot600

JesseDFBlue, 2018-05-10:
For BlHeli32 you can try this https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Rev32.41%20hex%20files, on a bench test it looks good.

degner, 2018-05-18:
I have the same problem as @oyvinla and @JesseDFBlue.  Helio Spring + Typhoon32 v2 4in1 running DSHOT600 16k/16k.  The current and voltage sensors do work but drop out often in flight giving alarming "Land Now" warnings etc.  I note @therealjonnyd suggests this is a known problem and to try DSHOT600 but I am already running this.  Can anyone say if we are waiting on a BLHeli32 update or BeF/BuF update?  And perhaps when it might arrive?

orneryd, 2018-06-07:
is this fixed in BeF?

degner, 2018-06-07:
> This is not fixed for me. Amp draw is way off and I can't calibrate it.

I am using a Helio Spring so I don’t think I can put Betaflight on it to
test?

IcyLemyur276 does the current change at all? I initaially did not
understand how to set the optional solder joins or software settings and
had a near constant current reading which could not be calibrated. Perhaps
this is your situation?

For me the voltage and current appears accurate most of the time but drops
to zero every now and then.

Tim if there is a binary you would like me to test I am happy to help.

David

IcyLemur276, 2018-06-07:
@degner yes that is the problem I’m having.  Current readings are constant/very high and don’t change based on throttle.  Can you suggest the solder joins and software setup to fix this? I’d be happy to email you my wiring/solder diagram.

Thetruesaint, 2018-08-07:
Same issue here. Helio + HW 4in1 Dshot600, When it is disarmed, it reads 7 or 8 Amps witch is not real.

Arcerys, 2018-08-25:
Where to connect on Helio FC the BAT pin from HW  4in1 harness?
I have connected HW current to FC rx4 pin, but have no voltage (0v avg cell and 5.2v total)
Power settings are Onboard ADC and no sensors configured on Ports page.

Imozeb, 2018-08-27:
Has anyone managed to solve this? I’m having the same issue with my Helios Spring FC and Dal RC engine.

Arcerys, 2018-08-27:
Solved my issue with HelioSpring V2 FC and HW 4in1 45A ESC.
Just powered FC from BAT pin on ESC, not from 5v and CRT pin on ESC to crnt sensor pad on FC.
In Butterflight set it as ”onboard adc” and that's it... Problem solved for me.

Imozeb, 2018-08-27:
@Arcerys How did you set it to onboard adc?

Also do you remember the ballpark for the current sensor offset and slope?

Thanks.

Imozeb, 2018-08-27:
Oh nevermind I found it. Are you using dshot600 or Multishot?

Arcerys, 2018-08-28:
Dshot1200

Imozeb, 2018-08-28:
@Arcerys Thanks. I tried Dshot1200 and Dshot600 and I still have the issue. 

Let me explain it further and maybe you guys can tell me if I missed anything. 

@IcyLemur276 BTW You have the same setup as me, maybe you have some insight?

I am using a Helios Spring FC and a DalRC 40A 4in1 both updated to the newest firmwares. I’ve tried dshot1200 and dshot600. I soldered and wired everything the same as IcyLemur276 post above.

I can see the amps on my OSD and it changes by the same amount when I change the throttle i.e increasing to half throttle increases amps by 15A always (I calibrated the slope).

The problem is that the offset changes when I shake or rotate my quadcopter. If it’s level, the idle (no spinning props) amps fluctuate around  4A-6A (it’s wrong but stable enough to calibrate. If I shake or tilt my quadcopter, the amps sometimes (seemingly randomly) jumps to fluctuate between 30A-50A. If I shake tilt it some more it returns to normal.

What is the most likely cause/fix for this? Is it a bad current shunt or a bad FC current ADC? Or is it EMI (the motors were not spinning).

Any advice would be appreciated. Thanks.

Imozeb, 2018-08-28:
Hi! I’ll try to explain it clearer. Let me know if I missed something that is important and I will fill in the information.

Background:
I’ve calibrated current sensors for my quadcopters before so I understand the procedure.

Problem:
If I tilt my quadcopter, the current sensor’s OSD reading jumps up 20+ Amps. This occurs when both armed and unarmed.

Specs:
Helios Spring Flight Controller using the current ADC and latest firmware

DalRC Engine 40A 4in1 ESC with built in current sensor and using latest firmware

5” quadcopter props

Emax 2207 2550Kv motors

Test 1:
I’ve tilted my quadcopter when armed and flying, when on the bench unarmed and motors not spinning, and when on the bench props off armed and motors spinning. At random angles, it changes the offset of the current sensor so the current OSD reading jumps up 20+ Amps. However, when I check the current offset in Butterflight it is still the same.

Test 2:
I have repeated test 1 running Multishot and dShot1200 and dShot600 with the same result.

Test 3:
I have calibrated the slope so the current OSD reading increases with increasing throttle. This proves that the current sensor is connected.

Secondly, it increases by the same amount based off throttle value. That is at half throttle it increases from the offset by X Amps. And at 3/4 throttle it increases from the offset by Y Amps. This is consistent no matter what the current offset is.

What do you think I can do to fix this changing current offset problem? And please let me know if you need more information.

dzenbot, 2018-11-05:
Same issue here :/