ButterFlight/butterflight #137: RC7-Caprica HELIOSPRING IMUF quaternion degradation

adrianmiriuta, 2018-04-27:
On longer flights (6-8 minutes)
HEADADJUST does not reset yaw drift to 0°
that means imufQuat is degraded  quaternionModulus(imufQuat ) != 1  (no longer a unit quaternion).
that is very bad !

adrianmiriuta, 2018-05-11:
@kidBrazil 
@orneryd 

I managed to flash the 
PR: IMUF quaternion degradation DEBUG #138
on a HELIOSPRING  with IMUF 106
and got the proof
the quaternionModulus(&qAttitude) value at flight end is 0.992  !
take a look to debug[0].
This is not OK !
do you want to normalize it in the IMUF code ?
or should  I supply a patch for  Butterflight 3.5.1 [ Release Candidate ]  ?

[image]

orneryd, 2018-06-07:
does this persist with Odin?

adrianmiriuta, 2018-06-07:
the quaternion degradation comes from the IMUF code(HELIOSPRING),
until now there is no fixed release!

closed 2018-10-05 by adrianmiriuta