ButterFlight/butterflight #152: EWRF e708TM3 VTX Not usable with OSD

Aigleblanc, 2018-05-17:
Hi,
 
In BTLF, active TBS smart audio in UART and is ok with OSD ( piggy osd v2 )
In BUTTER, active TBS smart audio reset config UART when i save config and reboot.

ref of vtx : https://www.banggood.com/EWRF-e7086TM3-5_8G-48CH-25mW200mW600mW-Switchable-FPV-Transmitter-Support-OSD-Configuring-p-1158300.html

Have a nive day.

Fabrice,

DrBonesLCD, 2018-06-04:
I have the same problem with the VTX EWRF  E709TM3 - I am using the Helio Spring FC, VTX smartaudio connected to UART 5 (TX5). In the OSD, it recognizes the vtx and displays Band/Channel/power used, but when trying to change any of that, it isn't implemented by the VTX.

https://www.banggood.com/fr/E709MT3-5_8G-40CH-25mW-200mW-600mW-Adjustable-AV-Transmitter-w-Mounting-Hole-for-Flight-Controller-p-1191064.html?utm_source=google&utm_medium=cpc_lang&utm_content=bruno&utm_campaign=fr-sds-all-test-5784&gclid=Cj0KCQjwxtPYBRD6ARIsAKs1XJ7dnlfdU1qZPU8J80b5WH81mteIpwnfwGjGUsL8gP42UDS_mg5G95saAmdBEALw_wcB

orneryd, 2018-06-07:
is this implemented in BeF?