ButterFlight/butterflight #153: copy and paste ...

robert-b, 2018-05-17:
https://github.com/ButterFlight/butterflight/blob/88adbaa3a5a6193353731bd6516e920e983be655/src/main/sensors/gyro.c#L1106

we need the single sample and not the rate change per microsecond.

adrianmiriuta, 2018-05-24:
@robert-b 
this code is correct, 
a change would destroy Attitude calculus.

Please take your time and think, 
or test before opening issues without reason.

robert-b, 2018-05-24:
surprise.

robert-b, 2018-05-24:
you need to take the sum of the samples and divide it by the sample count.
the code doesn't do anything to prevent drift.
yes i know it destroys the attitude.
fully flight tested: https://www.youtube.com/watch?v=TsI3Ag7eD0M

what does it destroy exactly?

adrianmiriuta, 2018-05-24:
@robert-b 
this has nothing to do with drift

you have 2 time domains
gyro loop (32k 16k ... xxx)
and attitude calculus (runs at 200 Hz)

You have to sum up the gyro increments and divide trough time 
to get the angle difference for attitude quaternion integration.

if you divide it trough sample count you get a sampling average,
not a accumulation average! 
(but you need a accumulation average ... absolute angle difference )

that's all .

adrianmiriuta, 2018-05-24:
@robert-b 
Iour FPV flight is acro-mode
the change you propose only affects stabilized flight modes
(which use Attitude quaternion).

robert-b, 2018-05-24:
it is not acro mode - just to prove the situation

robert-b, 2018-05-24:
and btw. your view on the units is strange.

adrianmiriuta, 2018-05-24:
@robert-b 
ok 
was nice talking to you ...
bye

robert-b, 2018-05-24:
what is your mother tongue?