ButterFlight/butterflight #158: BF 3.5.1 (RC2) freezes F4 CPU on OmnibusF4 corner by CLI command

jstremmler, 2018-05-26:
BF 3.5.1 (RC2) freezes F4 CPU on OmnibusF4 corner by CLI command.....

If I simply key in: "STATUS" as a CLI command the  F4 CPU freezes and board has to be unpowered and rebooted  again. 

This is not a hardware issue because it works with 3.5.0

adrianmiriuta, 2018-05-26:
@jstremmler 
use
Butterflight 3.5.1 [ Release Candidate ]
it's fixed there

itachi1983, 2018-05-27:
Same problem with omnibusf4FW 3.5.1 rc2, no issue with betaflight.

itachi1983, 2018-05-27:
But my fc is an airbot fireworks v2 , firmware is omnibusf4FW not SD

adrianmiriuta, 2018-05-27:
@itachi1983 
[butterflight_3.5.1_OMNIBUSF4FW.hex.zip](http://www.silverbug.org/butterflight_3.5.1_OMNIBUSF4FW.hex.zip)

itachi1983, 2018-05-27:
Solved cli problem but no oSD in configurator

itachi1983, 2018-05-27:
My foult problem seems solved cli commamds work and oSD too. Thanks

adrianmiriuta, 2018-05-27:
@itachi1983 
sorry
You have to wait for next release !
I fly LOS and have dropped the OSD out, 
because it fucks up the gyro data on this board the SPI3 is overloaded

orneryd, 2018-06-07:
is this merged into master @adrianmiriuta ?

adrianmiriuta, 2018-06-07:
no.

adrianmiriuta, 2018-06-30:
@orneryd 
you can close this it is merged in odin