ButterFlight/butterflight #161: Turtle Mode On Multishot

j33bs, 2018-05-28:
Can we please have a mode that switches to dshot and reverses props (from multishot) specifically for helio users looking for 32k pid loop. Sorry if this issue has already been listed, i couldn't find it.

Clynikal, 2018-06-10:
Thanks for posting this request, I think it’s a request to enable dshot on aux switch(s) for any dshot command that way we can still do beacon for failsafe in a mix.

Zuldan, 2018-06-11:
Turtle Mode On Multishot would be fantastic. @Clynikal thank you for looking into this. We appreciate it.

j33bs, 2018-06-11:
yeah if enabled on a switch i think their may need to be a change within the gui to allow flip over after crash to be assigned an aux channel while multishot is selected under configuration.

monko760, 2018-07-08:
Changing esc protocol usually requires the fc to be rebooted, doe sit not? Me thinks it wouldn't be to easy, but Im dumb and just wanted to say I WANT THIS FEATURE!

mirfpv, 2018-12-03:
upvote!

allanragec, 2019-11-02:
upvote