ButterFlight/butterflight #175: Racerstar Crazybee F3 Butterflight firmware no connection after flashing firmware

thomastt123, 2018-06-10:
I got a Crazybee F3 flightcontroller and flashed the firmware to spracingF3 version 3.5.0.
After this done succesfully there is no connection possible anymore through usb.
The connection port doesn't show up.

thomastt123, 2018-06-15:
I tried to get connection to the fc with windows 10.

I put the fc in dfu mode and reflashed it again to betaflight 3.3.0. 

With zadig I put the bootloader driver to winusb.

In standard mode (non bootloader mode), there isn't a port coming up though.

In windows I can get it to assign to COM3 when I put the usb driver to serial connection when the fc is in bootloader mode. 

The big problem seems to be that the fc doesn't put out a COM port connection signal.

Why I say this is because when connected in standard mode (not bootloader mode), in zadig there isn't a flightcontroller nor another "generic" usb device coming up.

thomastt123, 2018-06-15:
All betaflight drivers for vcp and 210 are installed correctly

githubDLG, 2018-06-16:
er....    have you check `List All Devices` in Options menu of Zaidg?  And even though, you still can not find the COM port of your FC?
By the way, there is another situation that will cause com port temporary disappear: gyro hardware error. You will need about 10~20 seconds to see the correct com port after fc finish it's initalize in this situation.

thomastt123, 2018-06-16:
I checked LIST ALL DEVICES and it's not there.
It also seems that the fc doesn't initialize like it normally should.
I will count the beeps and post in a moment

githubDLG, 2018-06-16:
haha , if the fc doesn't initialize like it normally should, then just wait for 10~20 seconds, then the com port will be founded~~
In this situation, there must be some hardware with gyro

thomastt123, 2018-06-17:
It doesn't show up after waiting for a long time.
I can flash it in DFU but when not in bootloader mode, there isn't a port showing up.
The number of beeps when plugging in a battery are: 3.
The 4th and 5th beep aren't there as should be.

githubDLG, 2018-06-21:
er... then i don't know what happened to your FC. Maybe consider to buy another new one is a better way? Haha...

ProjectMockingbird, 2018-07-13:
I JUST had that happen to mine when I was going from Betaflight 3.4 to the latest RC of Butterflight.  Now Windows say USB device not recognized.

frakkan, 2018-07-13:
Try ImpulseRc driver fixer.