ButterFlight/butterflight #177: SPRacing F3 Incompatible with 3.5.1-RC2

Opticulex, 2018-06-11:
The SPRacing F3 is incomaptible with 3.5.1-RC2 and, while still functioning, will potentially mute your flight controller from any USB communication with no fix. I've tried this on 2 old SPF3's and both of them have become unresponsive through USB and no amount of fixing will help. A big warning for SPF3 boards for the next RC would be good in case there are more issues.

thomastt123, 2018-06-17:
I have the same issue here.
After reflashing to betaflight 3.2.0 and putting the bootloader drivers (with zadig) to winusb.
I managed to get it to connect through usb but I don't find a suitable driver to get the flightcontroller "recognised" by windows.

thomastt123, 2018-06-17:
Do you know where I can find the microsoft usb serial device driver? It's not in my driver list on windows, others also don't work.

Opticulex, 2018-06-28:
I don't know about that driver. I had the STM32, Silabs CP2102 and WinUSB drivers and it registered the flight controller in Butterflight on COM3 but refused to connect/communicate or flash. I tried everything I could but assume the board is completely bricked. Sorry I couldn't help but this is a big issue so there should be some warning on it before flashing..

Zuldan, 2018-06-28:
Boot pins?

https://m.youtube.com/watch?v=Krv-CMSfb_M

Opticulex, 2018-07-05:
No its completely bricked.

Zuldan, 2018-07-05:
So F4 boot pins dont work? The firmware doesnt touch the boot loader code so its impossible to brick. Maybe you're not bridging the pins correctly.

iespoto, 2018-07-08:
Definitely you should be able to access STM32 chip directly via pinouts.

LexioTech, 2018-07-09:
Hi everyone, we will look into the issue as soon as possible, for the time being, if you have an STLINK, or you care to buy one for around 10$, you can re-flash the firmware using the data pins on your FC.

This tutorial will help you do that:
[(https://www.youtube.com/watch?v=Om6teCbJM4U)]

Basically, the STLINK will power up your FC (you need to bridge the boot pins or press the boot button when connecting it) and you can directly flash the FC using the data pin (apply the hex) without the configurator.

The STLINK is an all-around useful tool to have, so I highly recommend it. 

But all that said, we won't take any responsibility if you kill your FC by soldering to a bad pin, or doing something wrong. This is not an easy task, so only recommended if you have some confidence and knowledge in this area.

We will keep you guys posted on the progress of the fix, please be patient though :)

iespoto, 2018-07-09:
Brotip: if you are ordering stlink then also take some pin-clamper /
headers, no need to solder anything :)