ButterFlight/butterflight #180: Please add RevoltOSD target

gabeoravitz, 2018-06-13:
Exists for BeF. Non-OSD BuF works and flies on ROSD board except for OSD functionality.

mmakam2, 2018-06-14:
were you able to get 32k mode working with ROSD board with butterflight?

gabeoravitz, 2018-06-14:
Yes. It was flying pretty great actually.

mmakam2, 2018-06-14:
Odd I can't get mine to calibrate in 32k mode. I get arming code 12.

gabeoravitz, 2018-06-14:
Yea I see your issue in the Betaflight GitHub now. Kind of a lame response from mikeller. It works in FlightOne so it's likely not a 32k issue specific to the MPU. 

I'm not using a RevoltOSD though, I'm using a MillivoltOSD, but the targets all work for me for BeF and BuF.

mmakam2, 2018-06-14:
Yea mikeller kept closing my issues. Anyways I ended up ripping the revolt out and put in a heliorc spring. Now happily running 32k/32k.

gabeoravitz, 2018-06-14:
well glad you got it worked out. shame though because these RF boards are perfectly capable 32k boards that are pretty good quality, especially if you're the type that likes plugs instead of soldering. Not as good as true 32/32 but still.

mmakam2, 2018-06-14:
Hardware is excellent, software sucks. Flightone doesn't even work with crsf telemetry. The osd is not even close to as good as bf

gabeoravitz, 2018-06-14:
the software is a disgrace, honestly. I gave up on it a long time ago. Then last week I gave it a whirl again because PG3 said they had fixed some of the issues I had. Well I set it up and some of the stuff had been fixed. So I flew it, and it flew pretty good. So I went into the OSD menu after a quick test, changed some PIDs, then went to "Save" and as soon as I hit that, all 4 motors went to a million percent and the quad flung itself into my fence. Luckily it went away from me. They're saying it's a bug in BLHeli32, but like everything else, there is no issue with BuF and BeF. Never again. Unacceptable.

mmakam2, 2018-06-14:
Support is absolutely terrible too, and that's something they brag about. I tried chasing people down on discord and Facebook, noone responds and when they do, it's completely unhelpful. Even their exchange policy is a scam, you still have to pay for defective products..

gabeoravitz, 2018-06-14:
Yep.

varkokonyi, 2018-06-23:
@kidBrazil @orneryd Do you think it's possible to add a target for this lad?

gabeoravitz, 2018-06-25:
I would be so grateful!

kt4rj, 2018-07-30:
Yes please, we need a Revolt OSD target.