ButterFlight/butterflight #186: External beeper not working from Aux switch

neverknownbetter, 2018-06-23:
I have two quads. One with a betaflight3, the other omnibus F4 v5. Both quads the external beeper will not activate from the Aux switch allocated to the beeper. settings same on both quads.
# get beeper
beeper_inversion = OFF
Allowed values: OFF, ON

beeper_od = ON
Allowed values: OFF, ON

beeper_frequency = 0
Allowed range: 0 - 16000

beeper_dshot_beacon_tone = 0
Allowed range: 0 - 5

ledstrip_visual_beeper = OFF
Allowed values: OFF, ON

Opticulex, 2018-06-28:
Have you set it up in the modes tab?

neverknownbetter, 2018-06-29:
Yes set up in mode tab on both quads. Led flashing on fc when activated. Beeps when fc arms. No beep when mode tab turns on. Same with both fc's

neverknownbetter, 2018-06-29:
What's the resource for beeper, mine is B13 on both?