ButterFlight/butterflight #209: Spektrum 4649T in serial mode does not work on 3.5.0 to 3.5.2 rc3

seven2099, 2018-09-23:
Spektrum 4649T in serial mode does not work on 3.5.0 to 3.5.2 rc3

When connected to UART1's TX pin (working in all betaflight versions)
the 4649T receiver does not send input to the FC. 

Telemetry is received however.

**Flight controller configuration**
 ```serial 0 64 115200 57600 0 115200
feature RX_SERIAL
feature TELEMETRY
serialrx_provider = SRXL
serialrx_halfduplex = ON
map TAER1234 ```

**Setup / Versions**
 - OmnibusF4SD (NTXF4 version)
 - Integrated VTX - Tramp SA on UART3
 - Rest is run of the mill

seven2099, 2018-09-28:
is this dead?

butterSnake, 2018-10-03:
yep

seven2099, 2018-10-04:
as in you wont fix this or as in butterflight

Knjlexus, 2019-03-13:
I believe we put items we don't want stolen in safes. This software doesn't support that.