ButterFlight/butterflight #233: CPU load 50% increased

adrianmiriuta, 2018-10-27:
The update from 
butterflight 3.5.0 - 3.5.2 to  butterflight 3.6.0
amounts to a 50% CPU load increase , at standard settings !

Higher gyro, PID loop rates no longer possible !

JasonRodman, 2018-11-18:
I am experiencing same issue. Went from 3.5.2 to 3.6.2 and have to run 16k to keep the cpu load under control. Running the helio spring v1 board. Anyone else having same issue?

chester3, 2018-11-21:
Same problem here im running 16/16 now.

JasonRodman, 2018-11-21:
It turns out to be running dshot takes a lot more cpu than it used to. I can get it to work at 50% cpu running 32k gyro, 16k pid loop, dshot1200 with accelerometer off. In 3.5.2 I was able to run everything 32k, accellerometer on and dshot1200. Only way to get it to work full 32k is to switch to multishot, but you lose turtle mode. What changed?

closed 2019-06-02 by adrianmiriuta