ButterFlight/butterflight #247: Feature Request: Caddx Turtle UART support

Clynikal, 2018-12-19:
Caddx has a UART rx/rx protocol for camera control similar to the runcam protocol to enable start/stop recording etc that the BeF refuse to implement. 

Please check this to see if anything can be done with the caddx consultant: @Tiktiki
https://github.com/betaflight/betaflight/issues/6735

Tiktiki, 2018-12-20:
I am no longer working with Caddx.

Here is the current protocol documentation, it can't be changed or reworked because Caddx rebrands 3rd-party camera sensors and controllers and they don't have the source code to work with.

Good luck!

[Uart_Turtletv2.docx](https://github.com/ButterFlight/butterflight/files/2697061/Uart_Turtletv2.docx)

eijowk, 2019-01-24:
I have made a pull at INAV for a modified camera control supporting Caddx UART control.
I guess that it's pretty easy to move to Butterflight or Betaflight.