ButterFlight/butterflight #248: Arming Twice Binary 10 Strix

Bigjmf, 2018-12-23:
**Describe the bug**
A clear and concise description of what the bug is.
Current set is 
Hobbywing 45a ESC xrotor 4 in 1
Strix Binary 10 FC 
R9M Receiver 
Runcam mini DVR
Amax Inno 2308 1500kv 

ESC's arm twice on Butterflight 3.6.4

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.
This is my first time using Butterflight but arming twice is usually not normal should only are once

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
\\\
# version
# ButterFlight / STRIXF10 (SX10) 3.6.4 Dec 19 2018 / 02:30:59 (3ded515f3) MSP API: 1.40
# IMU-F Version: 110

board_name 
manufacturer_id 

# name
name JBird

# resources

# mixer

# servo

# servo mix

# feature
feature -AIRMODE

# map

# serial
serial 0 32 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1950 2075 0 0
aux 1 2 1 1925 2100 0 0
aux 2 13 6 1400 1600 0 0
aux 3 28 1 1400 1600 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 1
set imuf_mode = 32
set imuf_rate = 32K
set imuf_w = 64
set acc_calibration = 60,-24,-54
set mag_declination = -11000
set min_check = 1015
set max_check = 1995
set rssi_channel = 16
set rssi_scale = 110
set serialrx_inverted = ON
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set vbat_full_cell_voltage = 42
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set pid_process_denom = 2
set tlm_inverted = ON
set osd_units = IMPERIAL
set osd_rssi_alarm = 30
set osd_vbat_pos = 2424
set osd_rssi_pos = 2073
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2049
set osd_craft_name_pos = 2060
set osd_warnings_pos = 2442
set vtx_band = 5
set vtx_channel = 7
set vtx_freq = 5880
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 3000

# rateprofile
rateprofile 0

set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 80
set tpa_rate = 15
set tpa_breakpoint = 1500

# 
\\\
**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

Hobbywing 45a ESC xrotor 4 in 1 (Bought Heli Direct)
Strix Binary 10 FC  (Bought Heli Nation)
R9M Receiver (Bought Heli Nation)
Runcam mini DVR (Bought Heli Nation)
Amax Inno 2308 1500kv (Amax Shop)
Unify Pro HV VTX 
**Additional context**
Add any other context about the problem here.

bzinner, 2019-01-08:
I'm having the same issue. Strix f10 + typhoon32 4 in 1 esc. Also,  bz+bz- and led pads not working/ configured( ground not being switched). The dshot beacon tone slider in butterflight configurator turns itself on and off if you leave the config tab and come back to it. If you enable dshot beacon and save,  it disables after reboot like there's a conflict. The beeper doesn't not show up in the resource list or features list in cli. It seems like the FC pins were never configured.  I have a 2nd (new) Strix f10  with nothing connected and has the same issue.  Tried 3.6.2 on both boards as well.

bzinner, 2019-01-09:
I didn't find a solution in 3.6.4, but flashed back to 3.6.2 again and confirmed the esc issue went away. I'm using Dshot1200 and also noticed while running 3.6.4, the ESCs were showing "bad frames" in Blheli32 configurator. The startup tones would be erratic at times.  These issues also went away after flashing back to 3.6.2
Still haven't resolved the beeper issue....

bzinner, 2019-01-09:
Ok.  I'm running Blheli32 rev 32.6 on the ESCs. Not sure if that matters.  Seems like a config issue on the controller

bzinner, 2019-01-09:
I was able to contact someone at Ready Made RC. Apparently it is a known issue they are working on. They also replied here  in the comments:

https://m.youtube.com/watch?v=nnPefQ4EZrA

MrNoizeFPV, 2019-01-09:
If you are seeing double beeps with the esc try flashing back to 32.4 firmware in blheli a lot of people have noticed double esc tones on 32.6 esc firmware, flashing back gets rid of that.

Bigjmf, 2019-01-09:
Ah perfect I had also contacted them about it and they hadn't responded
yet. That's when I then posted on github. It's cool though anything new
coming out in this hobby always has bugs. I pulled mine out for now and
went back to my trusted CL-racing board for now. Going to wait till they
work the bugs out and put on betaflight. I don't know if I'm just used to
it or I actually like it better. But I did not really care for the
butterflight over betaflight.

MrNoizeFPV, 2019-01-09:
yeah the double esc tones is a blheli issue not the strix board

Bigjmf, 2019-01-09:
Thanks I didn't even think about that angle. Appreciate your help.