ButterFlight/butterflight #249: Matek F405 3.6 -  Setpoint Weight/2nd Filters not saving!!!

Opticulex, 2019-01-19:
It happened in previous versions and apparently it's still happening now like 6 months later. My quad files like shit and there's nothing I can do about it. I don't really feel like putting an older version on my board as it will erase my complex config so I hope this can be fixed in the next update.

fxbisto, 2019-01-20:
This was happening to mine a while back, my workaround was to set everything up in the CLI instead. Could you post the results of _get gyro_ in the CLI? Here's mine for reference. flies great on a really noisy setup:
`# get gyro
align_gyro = DEFAULT
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

gyro_hardware_lpf = NORMAL
Allowed values: NORMAL, EXPERIMENTAL, 1KHZ_SAMPLING

gyro_sync_denom = 2
Allowed range: 1 - 32
Default value: 1

gyro_lowpass_type = KALMAN
Allowed values: PT1, BIQUAD, KALMAN

gyro_lowpass_hz = 100
Allowed range: 0 - 16000
Default value: 0

gyro_lowpass2_type = KALMAN
Allowed values: PT1, BIQUAD, KALMAN
Default value: BIQUAD

gyro_lowpass2_hz = 100
Allowed range: 0 - 16000
Default value: 90

gyro_notch1_hz = 400
Allowed range: 0 - 16000
Default value: 0

gyro_notch1_cutoff = 300
Allowed range: 0 - 16000
Default value: 0

gyro_notch2_hz = 200
Allowed range: 0 - 16000
Default value: 0

gyro_notch2_cutoff = 100
Allowed range: 0 - 16000
Default value: 0

gyro_calib_duration = 125
Allowed range: 50 - 3000

gyro_calib_noise_limit = 48
Allowed range: 0 - 200

gyro_offset_yaw = 0
Allowed range: -1000 - 1000

gyro_filter_q = 3200
Allowed range: 0 - 16000
Default value: 400

`

fxbisto, 2019-01-20:
I think it's a bug with the configurator rather than the firmware, I have the same thing with mine. Just use the CLI to configure it, I know it's less straightforward.
You said it flies poorly, can you post up the output of _get gyro_? Mine flies quite well on an F3 board with really noisy motors and a 3D printed frame. Check my settings for reference:
```
# get gyro
align_gyro = DEFAULT
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

gyro_hardware_lpf = NORMAL
Allowed values: NORMAL, EXPERIMENTAL, 1KHZ_SAMPLING

gyro_sync_denom = 2
Allowed range: 1 - 32
Default value: 1

gyro_lowpass_type = KALMAN
Allowed values: PT1, BIQUAD, KALMAN

gyro_lowpass_hz = 100
Allowed range: 0 - 16000
Default value: 0

gyro_lowpass2_type = KALMAN
Allowed values: PT1, BIQUAD, KALMAN
Default value: BIQUAD

gyro_lowpass2_hz = 100
Allowed range: 0 - 16000
Default value: 90

gyro_notch1_hz = 400
Allowed range: 0 - 16000
Default value: 0

gyro_notch1_cutoff = 300
Allowed range: 0 - 16000
Default value: 0

gyro_notch2_hz = 200
Allowed range: 0 - 16000
Default value: 0

gyro_notch2_cutoff = 100
Allowed range: 0 - 16000
Default value: 0

gyro_calib_duration = 125
Allowed range: 50 - 3000

gyro_calib_noise_limit = 48
Allowed range: 0 - 200

gyro_offset_yaw = 0
Allowed range: -1000 - 1000

gyro_filter_q = 3200
Allowed range: 0 - 16000
Default value: 400

```