ButterFlight/butterflight #252: strixf10 beeper wont work from aux and curr pad doesnt show reading

JustinBluntFPV, 2019-02-28:
I cant find which pin the beeper is resourced to and the curr pad wont show a reading when connected to my Matek XT60 w/ external current sensor. resource list shows adc_curr 1 a0,  but when i resource it to pin a01 and then save, it doesnt save/keep.  unfortunately its a strixf10 board running the latest butterflight and IMU-F.

lorentzm, 2019-03-12:
I had a look at this today since I'm still annoyed that the beeper doesn't work. It was a super simple fix, I'll copy the diff below. However, I don't have access to push a branch and open a pull request on the repo.

diff --git a/src/main/target/STRIXF10/target.h b/src/main/target/STRIXF10/target.h
index e5ceec458..d05e012be 100644
--- a/src/main/target/STRIXF10/target.h
+++ b/src/main/target/STRIXF10/target.h
@@ -22,7 +22,8 @@
 
  #define LED0_PIN                PB7
 
- #define BEEPER                  PC15
+ #define USE_BEEPER
+ #define BEEPER_PIN              PC15
  #define BEEPER_INVERTED
 
  #define USE_GYRO

daehani79, 2019-03-13:
@lorentzm Perfectly work!! Thank you!

mmakam2, 2019-03-13:
I am also having issues with the current pad. Not sure how to get it to work, tried everything, If anyone has gotten it to work, please share config.

lorentzm, 2019-03-13:
The current pad was working for me. I was using an Aikon AK 32 4-in-1 which for some reason has noise on the current output when using DSHOT but not when using ONESHOT. So, I wouldn't get anything useful for current when using DSHOT but got correct readings with ONESHOT.  I was able to clean up the noise and use DSHOT by soldering a 10000pf 50V cap between the current pad and the ground on the STRIXF10 board.

mmakam2, 2019-03-13:
can you please share your config?

mmakam2, 2019-03-13:
I'm seeing activity on onboard adc.even when esc is disconnected from current pad, am I supposed to use that or msp sensor?

lorentzm, 2019-03-13:
I killed my board putting it in a different quad yesterday :( But current was working from the current pad by using "Onboard ADC".

mmakam2, 2019-03-14:
ok thanks, did you have to remap any resources?

lorentzm, 2019-03-14:
No resources needed remapping.