ButterFlight/butterflight #253: Kalman filter (full dynamic implementation)

adrianmiriuta, 2019-03-06:
@apocolipse
@kidBrazil 
@LexioTech 

Hi,
I am working on a clean, full dynamic, implementation of the Kalman filter into Butterflight,
using the gpl released IMU-F code from heliorc.

anyone interested ?

kidBrazil, 2019-03-06:
That would be pretty awesome! Have you taken a look at Pegasus yet?

adrianmiriuta, 2019-03-07:
@kidBrazil  not yet (I am a cli user)

I already submitted 
PR #237  #244  #251 
but there seems to be no active replay from anyone.

I would like to go forward with ButterFlight development,
I would like to get merge rights to the project.

butterSnake, 2019-03-08:
[image]

robert-b, 2019-03-08:
i am planning to add a dynamic fft filter.
just a few tabs in the lower frequency band.

adrianmiriuta, 2019-03-08:
@butterSnake 
awesome, at last someone alive ...

robert-b, 2019-03-08:
you shouldn't use the word kalman ;-)
i haven't checked the mcu runtime for the dynamic thing.
but a simpler calculation using stdev calculation might do the trick as well.

butterSnake, 2019-03-08:
Ohh no the return of the mad one 😂

robert-b, 2019-03-08:
you should face it.
even 32k is dead ;-)

butterSnake, 2019-03-08:
It’s ok 32k is a dead end 🤷‍♂️

adrianmiriuta, 2019-03-08:
@robert-b 
it is Kalman (or Markov if you want ...)
I have a testing draft.
It is the helio IMU-F implementation without cross axes measurement noise variance.
On a CLRACING F7 it needs 1% more CPU (at 8Khz Gyro and PID loop).

robert-b, 2019-03-08:
32k is the mechanical resolution.
a ride on a gear wheel makes no sense.

adrianmiriuta, 2019-03-08:
@robert-b 
what do you mean with mechanical resolution ?

The MEMS operating frequencyes are 
X = 33+/-3 kHz, Y = 30+/-3 kHz, and Z = 27+/-3 kHz

and the maximal Output Data Rate for the ICM series is 8K . 
there is no 32K ...

robert-b, 2019-03-08:
mechanical resolution is the mechanical resolution.
now you have some sort of mcu inside.
would you dynamically adjust the reading frequency?

33k +-3k is 32k
which will prob. change with every production run.
as it did happen with the analog variants.

the word mechanical resolution was used and introduced by a tdk man.

adrianmiriuta, 2019-03-08:
@robert-b 

mechanical resolution of what ?

the MEMS operating frequency has nothing to do with Output Data Rate.

iggut, 2019-04-08:
I am glad to see there might be more to come for Butterflight. I love my helio boards and wish I had the skills to help out here hehe 
Thanks for all that do the work!

iggut, 2019-04-08:
I am no programmer or even sure I understand this so bear with me.

Would it not be a good idea to try using the BF4 rpm filter, but run it on the IMU-F or some sort of implementation of the dshot bidirectional communication to get RPM in real time for filter use (unsure how this part works)

adrianmiriuta, 2019-04-09:
@iggut 
this entire gyro filtering mania is just a band-aid, 
and has taken ridiculous proportions (36 notch filters).
All this band-aids try to fix the reality that a PID is not suitable for controlling highly nonlinear systems.
All this filtering is killing the dynamic of the quad, you get the best handling with lowest filtering.
It is crucial to pick a low noise frame-esc-motor-prop combination, with less filtering it will fly the best.

butterSnake, 2019-04-09:
Lols do you have any idea what kinda total latency you can achieve with the rpm filtering? Same old bull shit from the same old bunch 😂 @iggut such a thing an’t ever gona happen round here as it requires dshot to be used for starters and frankly it’s way above any ones pay grade.. this project is dead as a door nail 😂

adrianmiriuta, 2019-04-10:
@butterSnake 
Thx for your advice, but there is always another way to do things.
You can use what U like it's all GPL.
But please stop the trolling,  you're not in the kindergarten.

adrianmiriuta, 2019-04-22:
@apocolipse
@kidBrazil
@LexioTech

PR with Kalman implementation is ready.
some smaller F3  do not have enough RAM for it.

theSkonk, 2019-05-05:
@adrianmiriuta  its was good enough for the butter devs at the beginning of this project... love how quickly people forget their own  shit behaviour...  good to see this project is a massive dead end still :P

robert-b, 2019-05-05:
not true.
but pls. have a look at the code conduct at the beta fligth site.
in the mean time i know you a bit better.

closed 2019-06-02 by adrianmiriuta

theSkonk, 2019-06-03:
aaa

robert-b, 2019-06-08:
dear skonk,
my thinking says iggut is right.
what you love is a piece of over engineered software.
when you state that a filter has no phase delay then this is the biggest bullshit i've ever read.

Quick-Flash, 2019-09-27:
@adrianmiriuta have you heard about the Emuflight project?